acmerobotics/road-runner-quickstart #9: OpModes have unexpected behavior

tiborr, 2018-11-09:
When using the DriveFFTuningOpMode with a SampleMecanumDriveREV base, it moves less than a foot forward (and also turns slightly). The phone telemetry suggests it should move closer to 100 in. forward - is this unexpected behavior/do you know why this might be? If so, it would suggest that the kV (and kA/kStatic) that are given are incorrect. Subsequently, the StraightTestOpMode drives at about a 10 degree angle instead of straight and seems to get more off-track as time progresses. I've set my other DriveConstants accurately (I believe). I'm also using RevRobotics20HdHexMotor. Thank you!

guberti, 2018-11-09:
Did you change what motors you were using in your `SampleMecanumDriveREV` class? `DriveFFTuningOpMode` works fine for me (I'm using Orbital 20s geared 1:1)

tiborr, 2018-11-10:
Do you mean changing the deviceName parameter when getting the motors from the hardware map? If so, then yes I changed that and also set the directions of the motors.

tiborr, 2018-11-12:
Do you need to tune the PID coefficients for the StraightTestOpMode and/or for the DriveFFTuningOpMode?

rbrott, 2018-11-13:
Tuning the **follower** PID coefficients is not necessary for either of those op modes. In fact, I discourage people from adding feedback too early as it's crucial that the feedforward response is good (e.g., there isn't a bad motor parameter, wheel radius). On the other hand, tuning the **drive velocity** PID can be be a good idea, although it is similarly not required for the routines you mentioned.

To diagnose your problem, I recommend double checking all of the values in `DriveConstants`. A slight drift is somewhat acceptable in the early tuning phases (it doesn't affect the coefficient determination very much), and it will be addressed by feedback at a later stage. However, if the drift is out of control, I suspect there may be core hardware issues that require immediate attention.

rbrott, 2018-11-21:
@tiborr Are you still having problems?

closed 2018-11-28 by rbrott