acmerobotics/road-runner-quickstart #13: Robot won't move in StraightTestOpMode

xRoALex, 2019-01-03:
Our robot move only in the tuning OpModes, but not in the test ones. We have set the hardware names and imu.

rbrott, 2019-01-03:
Is your code available somewhere? In order for the robot to move in `StraightTestOpMode`, `DriveConstants.kV` must be set.

xRoALex, 2019-01-04:
I have made it work. Now i have some other questions. If I set the kv and track width so that the StraightTest and TurnTest work close to perfectly, the SplineTest will have a big offset. And if I set the parameters for the SplineTest the other two will have the big offset. How can i correct it? Can it be corrected with PID tunning? If so, how do i use those graphs in the dashboard? What do I have to look for?

rbrott, 2019-01-04:
I need a bit more information to properly diagnose this issue. What kV and track width values are you using for both? Are you using the built-in velocity PID (i.e., `RUN_USING_ENCODER`)? What drivetrain are you using? Are all of the measurements in `DriveConstants`? What's the qualitative behavior of the robot when there's a "big offset"?

In general, consistent errors with high magnitude are due to bad feedforward parameters. If the errors are more random or smaller magnitude then feedback (e.g., PID) can mostly eliminate them. **I strongly caution against prematurely adding feedback; it's important to get the feedforward-only/open-loop response as close as possible first.

xRoALex, 2019-01-04:
I am using kV = 0.2696  and TrackWidth = 14.95, built-in PID and mecanum drivetrain. In DriveConstants i have set everything except kA and kStatic. The offset is half a tile less on x-axis and half a tile more on y-axis in the SplineTest with these kV and TrackWidth values, and the StraightTest, TurnTest and FollowerPIDTest work fine. I should mention that I have tweaked these two values a little bit so that those 3 test work almost perfectly.

rbrott, 2019-01-05:
That kV value is a little high but still plausible. If you're sure about the constants, I'd try reducing the maximum speed and acceleration of the spline. That should give a good indication of whether the wheels are slipping too much, the robot is over-rotating, etc. If there are still problems at lower speeds, it'd be great if you could post a video of the response. Additionally, try printing/logging/telemetry-ing your error, especially at the end of the move, and see if it matches the actual error.

xRoALex, 2019-01-05:
Everything is fine now. I have one more question though. Is there anyway we can pause the trajectory for putting down the team marker, or do I have to make 2 separate trajectories? Anyway, thank you a lot for your help and quick responses, and for the amazing job you have done with this API.

rbrott, 2019-01-05:
I would probably make two separate trajectories in that case. You _can_ add a wait segment although it can be tricky to time the concurrent actions correctly. Glad you got things working :+1:

closed 2019-01-05 by xRoALex