acmerobotics/road-runner-quickstart #21: Changing distance on dashboard for DriveVelocityPIDTuner makes no difference

alan412, 2019-09-21:
I don't know if we are doing something really bizarre or not but with the DriveVelocityPIDTuner changing the distance on the dashboard makes no difference.  (but changing it in the code does)

We are hesitant to tune the PID values if this makes no difference.

rbrott, 2019-09-21:
That seems a bit strange. Do you have the same problem with other dashboard variables? Note that changes to `DISTANCE` won't be reflected until the start of the next motion.

alan412, 2019-09-21:
Don't know.  we didn't try others.  The changes didn't take effect when we changed it after pressing stop and then pressing start again.

Here is all our code in case we changed something weird: https://github.com/ftc16072/roadrunner-quickstart
(but I did a git diff with the original and I can't see anything...)

rbrott, 2019-09-21:
Did you save the modified values (either by pressing enter or clicking the save/floppy icon in the top right)?

alan412, 2019-09-22:
You have got to be kidding me.   I didn’t know you had to do that.  We spent HOURS on this.  Feeling really stupid now.

closed 2019-09-22 by rbrott