acmerobotics/road-runner-quickstart #22: FTCDashboard: Infinity is not a valid double value as per JSON specification.

Calvin-Xu, 2019-10-01:
This happens when an opMode using FTCDashboard is run:

Stacktrace:
`E/fi.iki.elonen.NanoHTTPD: Communication with the client broken, or an bug in the handler code
    java.lang.IllegalArgumentException: Infinity is not a valid double value as per JSON specification. To override this behavior, use GsonBuilder.serializeSpecialFloatingPointValues() method.
        at com.google.gson.Gson.checkValidFloatingPoint(Gson.java:324)
        at com.google.gson.Gson$2.write(Gson.java:292)
        at com.google.gson.Gson$2.write(Gson.java:278)
        at com.google.gson.Gson.toJson(Gson.java:669)
        at com.google.gson.Gson.toJsonTree(Gson.java:562)
        at com.google.gson.Gson.toJsonTree(Gson.java:541)
        at com.google.gson.internal.bind.TreeTypeAdapter$GsonContextImpl.serialize(TreeTypeAdapter.java:155)
        at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:22)
        at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:12)
        at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:81)
        at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:74)
        at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
        at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
        at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
        at com.google.gson.Gson.toJson(Gson.java:669)
        at com.google.gson.Gson.toJsonTree(Gson.java:562)
        at com.google.gson.Gson.toJsonTree(Gson.java:541)
        at com.google.gson.internal.bind.TreeTypeAdapter$GsonContextImpl.serialize(TreeTypeAdapter.java:155)
        at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:22)
        at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:12)
        at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:81)
        at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:74)
        at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
        at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
        at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
        at com.google.gson.Gson.toJson(Gson.java:669)
        at com.google.gson.Gson.toJsonTree(Gson.java:562)
        at com.google.gson.Gson.toJsonTree(Gson.java:541)
        at com.google.gson.internal.bind.TreeTypeAdapter$GsonContextImpl.serialize(TreeTypeAdapter.java:155)
        at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:22)
        at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:12)
        at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:81)
        at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:74)
        at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
        at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.write(ReflectiveTypeAdapterFactory.java:125)
        at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.write(ReflectiveTypeAdapterFactory.java:243)
        at com.google.gson.Gson.toJson(Gson.java:669)
        at com.google.gson.Gson.toJson(Gson.java:648)
        at com.google.gson.Gson.toJson(Gson.java:603)
        at com.google.gson.Gson.toJson(Gson.java:583)
        at com.acmerobotics.dashboard.FtcDashboard.toJson(FtcDashboard.java:612)
        at com.acmerobotics.dashboard.DashboardWebSocket.send(DashboardWebSocket.java:72)
        at com.acmerobotics.dashboard.FtcDashboard.addSocket(FtcDashboard.java:814)
        at com.acmerobotics.dashboard.DashboardWebSocket.onOpen(DashboardWebSocket.java:34)
        at fi.iki.elonen.NanoWSD$WebSocket$1.send(NanoWSD.java:87)
        at fi.iki.elonen.NanoHTTPD$HTTPSession.execute(NanoHTTPD.java:964)
        at fi.iki.elonen.NanoHTTPD$ClientHandler.run(NanoHTTPD.java:192)
    	at java.lang.Thread.run(Thr
E/EventLoopManager: Event loop threw an exception
E/EventLoopManager: java.lang.IllegalArgumentException: Infinity is not a valid double value as per JSON specification. To override this behavior, use GsonBuilder.serializeSpecialFloatingPointValues() method.
E/EventLoopManager:     at com.google.gson.Gson.checkValidFloatingPoint(Gson.java:324)
E/EventLoopManager:     at com.google.gson.Gson$2.write(Gson.java:292)
E/EventLoopManager:     at com.google.gson.Gson$2.write(Gson.java:278)
E/EventLoopManager:     at com.google.gson.Gson.toJson(Gson.java:669)
E/EventLoopManager:     at com.google.gson.Gson.toJsonTree(Gson.java:562)
E/EventLoopManager:     at com.google.gson.Gson.toJsonTree(Gson.java:541)
E/EventLoopManager:     at com.google.gson.internal.bind.TreeTypeAdapter$GsonContextImpl.serialize(TreeTypeAdapter.java:155)
E/EventLoopManager:     at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:22)
E/EventLoopManager:     at com.acmerobotics.dashboard.config.variable.ConfigVariableSerializer.serialize(ConfigVariableSerializer.java:12)
E/EventLoopManager:     at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:81)
E/EventLoopManager:     at com.google.gson.internal.bind.TreeTypeAdapter.write(TreeTypeAdapter.java:74)
E/EventLoopManager:     at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
E/EventLoopManager:     at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:208)
E/EventLoopManager:     at com.google.gson.internal.bind.MapTypeAdapterFactory$Adapter.write(MapTypeAdapterFactory.java:145)
E/EventLoopManager:     at com.google.gson.Gson.toJson(Gson.java:669)
[... 18 lines cut ...]
        at com.google.gson.internal.bind.TypeAdapterRuntimeTypeWrapper.write(TypeAdapterRuntimeTypeWrapper.java:69)
E/EventLoopManager:     at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$1.write(ReflectiveTypeAdapterFactory.java:125)
E/EventLoopManager:     at com.google.gson.internal.bind.ReflectiveTypeAdapterFactory$Adapter.write(ReflectiveTypeAdapterFactory.java:243)
        at com.google.gson.Gson.toJson(Gson.java:669)
E/EventLoopManager:     at com.google.gson.Gson.toJson(Gson.java:648)
        at com.google.gson.Gson.toJson(Gson.java:603)
E/EventLoopManager:     at com.google.gson.Gson.toJson(Gson.java:583)
E/EventLoopManager:     at com.acmerobotics.dashboard.FtcDashboard.toJson(FtcDashboard.java:612)
        at com.acmerobotics.dashboard.DashboardWebSocket.send(DashboardWebSocket.java:72)
E/EventLoopManager:     at com.acmerobotics.dashboard.FtcDashboard.sendAll(FtcDashboard.java:804)
E/EventLoopManager:     at com.acmerobotics.dashboard.FtcDashboard.updateConfig(FtcDashboard.java:659)
E/EventLoopManager:     at com.acmerobotics.dashboard.FtcDashboard.onOpModePostStop(FtcDashboard.java:916)
        at org.firstinspires.ftc.robotcore.internal.opmode.OpModeManagerImpl.callActiveOpModeStop(OpModeManagerImpl.java:402)
E/EventLoopManager:     at org.firstinspires.ftc.robotcore.internal.opmode.OpModeManagerImpl.runActiveOpMode(OpModeManagerImpl.java:327)
E/EventLoopManager:     at com.qualcomm.ftccommon.FtcEventLoop.loop(FtcEventLoop.java:214)
E/EventLoopManager:     at com.qualcomm.robotcore.eventloop.EventLoopManager$EventLoopRunnable$1.run(EventLoopManager.java:301)
E/EventLoopManager:     at com.qualcomm.robotcore.util.ThreadPool.logThreadLifeCycle(ThreadPool.java:737)
E/EventLoopManager:     at com.qualcomm.robotcore.eventloop.EventLoopManager$EventLoopRunnable.run(EventLoopManager.java:244)
E/EventLoopManager:     at com.qualcomm.robotcore.eventloop.EventLoopManager$1.run(EventLoopManager.java:569)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1162)
E/EventLoopManager:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:636)
        at com.qualcomm.robotcore.util.ThreadPool$ThreadFactoryImpl$1.run(ThreadPool.java:793)
E/EventLoopManager:     at java.lang.Thread.run(Thread.java:764)
V/RobotCore: RobotCoreException in EventLoopManager: EventLoop Exception in loop(): IllegalArgumentException - Infinity is not a valid double value as per JSON specification. To override this behavior, use GsonBuilder.serializeSpecialFloatingPointValues() method.
    EventLoopManager state is EMERGENCY_STOP`

rbrott, 2019-10-01:
This is the intended behavior of the dashboard (the web UI isn't equipped to edit these kind of double arguments). If you'd like this functionality to be added, please open an issue in the dashboard project.

That said, if you think there's an issue with the quickstart putting an infinite in a dashboard configuration variable, can you give some more context (e.g., what tuning program you were running, your modifications to the quickstart code)?

Calvin-Xu, 2019-10-01:
Thank you! I do not think we have not changed any configurations for the dashboard ourselves. Attached are the files we have changed (`SampleMecanumDriveREVOptimized`, `DriveConstants` and `DriveVelocityPIDTuner`).

[changedFiles.zip](https://github.com/acmerobotics/road-runner-quickstart/files/3678911/changedFiles.zip)

Unless...we should not set `drive = new SampleMecanumDriveREVOptimized(hardwareMap);` in `DriveVelocityPIDTuner`, but keep the base class `SampleMecanumDriveBase`?

rbrott, 2019-10-02:
Oh I think I see the issue. In `DriveConstants.java` you (likely unintentionally) have a zero `GEAR_RATIO` so `kV` is infinity from a divide by zero. To fix this, change the ratio from `1/20` to `1.0/20.0`. (side note: that seems like a strange gear ratio; make sure you're ignoring the initial ratio of the motor gearbox)

Calvin-Xu, 2019-10-02:
Thank you! We did make mistakes with both integer division behavior and the configuration itself (counting the gearbox ratio).

So if the gear on the driving wheel is 16 teeth, and on the motor shaft 32, then `GEAR RATIO` should be `// output (wheel) speed / input (motor) speed` which is inverse to the number of teeth and therefore `32.0/16.0`?

rbrott, 2019-10-02:
Yes, that's correct.

Calvin-Xu, 2019-10-02:
Thank you for the generous support. We can confirm FTCDashboard is now working.

closed 2019-10-02 by Calvin-Xu