acmerobotics/road-runner-quickstart #23: [Q] Same PID coefficients for all motors on the drivetrain?

Calvin-Xu, 2019-10-03:
We are wondering if there is a reason that the same PID coefficients are set for all motors on the drivetrain (for example `setPIDCoefficients` in `SampleMecanumDriveREVOptimized`). It seems to us that on our drivetrain each motor/wheel has a different error and should be tuned separately?

rbrott, 2019-10-03:
For common drive configurations, the load on each motor should be pretty much the same, so the same coefficients should work for all of them. Keep in mind that each motor has its own control loop and error even if the coefficients are the same. If you have a particularly skewed weight distribution or a special hardware setup, you can definitely tune the motor coefficients separately.

Calvin-Xu, 2019-10-03:
Thank you for the clarification!

closed 2019-10-03 by Calvin-Xu