acmerobotics/road-runner-quickstart #24: [Q] Pose2D heading and IMU raw heading

Calvin-Xu, 2019-10-12:
For `TrackWidthTuner`, we are trying to implement a localizer that updates only heading using the internal REV IMU (the `imu` initialized in all sample mecanum drive subclasses). 

We read [here](https://acme-robotics.gitbook.io/road-runner/tour/coordinate-frame) that:

> the robot has a heading  defined as the angle \theta between the front of the robot and the global  axis

And implemented the following logic in our localizer:

```java
public class NaiveHeadingOnlyLocalizer implements Localizer {

    private Pose2d poseEstimate;
    private SampleMecanumDriveREVOptimized drive;

    public NaiveHeadingOnlyLocalizer(SampleMecanumDriveREVOptimized drive) {
        this.drive = drive;
    }

    @NonNull
    @Override
    public Pose2d getPoseEstimate() {
        return poseEstimate;
    }

    @Override
    public void setPoseEstimate(@NonNull Pose2d pose2d) {
        this.poseEstimate = pose2d;
    }

    @Override
    public void update() {
        poseEstimate = new Pose2d(0.0,0.0, getProcessedHeading());
    }

    public double getProcessedHeading() {
        if (drive.getRawExternalHeading() < 0) {
            return 2 * Math.PI + drive.getRawExternalHeading();
        } else {
            return drive.getRawExternalHeading();
        }
    }
}
```

The headings do show up as we expect them to be on the dashboard. However, the effective track width returned seems very off, and we wonder if we have understood something wrong. Any help is appreciated!

rbrott, 2019-10-12:
First of all, what's the motivation for using a new localizer in the first place? It seems like one of the standard drive wheel localizers (e.g., `MecanumDrive.MecanumLocalizer`) would work fine.

If you're still intent on using a custom localizer, it's better to use `Drive#getExternalHeading()` instead of `Drive#getRawExternalHeading()`. These values can be used directly; there's no need to conditionally add `2 * Math.PI` as in `getProcessedHeading()`.

Calvin-Xu, 2019-10-12:
Thank you. We did not realize `MecanumDrive` has a default localizer. Would you explain the relationship between `MecanumDrive.MecanumLocalizer` and `StandardTrackingWheelLocalizer` in `localizer`?

rbrott, 2019-10-12:
Sure! `MecanumDrive.MecanumLocalizer` uses the encoders on the drive motors and (by default) the IMU for odometry. Every `MecanumDrive` instance comes with this `Localizer` by default (and there are analogous `Localizers for other drive types). `StandardTrackingWheelLocalizer` is a common implementation of `ThreeTrackingWheelLocalizer`. This class (and its sibling `TwoTrackingWheelLocalizer`) are intended for external tracking/dead/omni wheel odometry (which has become the de facto standard in FTC).

Calvin-Xu, 2019-10-12:
Thanks! This is very helpful.

closed 2019-10-12 by Calvin-Xu