acmerobotics/road-runner-quickstart #27: Problems finding ExpansionHubEx

EddieDL, 2019-10-26:
I am not sure this is the right place to ask this. If it is not, I will post to one of the other GitHub repos for the applicable related library.

It seems like the code to retrieve the expansionhub is "flaky". If we deploy new code, the app crashes, etc. you can no longer find the expansionhub unless you restart the robot and app. 

Is there something we are doing wrong or is it just the cost of being able to do "bulk reads"?

rbrott, 2019-10-26:
This is an issue with https://github.com/OpenFTC/RevExtensions2

closed 2019-10-26 by rbrott

rbrott, 2019-10-26:
You can try downgrading RE2 to 1.1 (i.e., replace `implementation 'org.openftc:rev-extensions-2:1.2'` with `implementation 'org.openftc:rev-extensions-2:1.1'` in `TeamCode/build.release.gradle`) and add `RevExtensions2.init()` to the beginning of the op modes.

EddieDL, 2019-11-13:
This is an interesting idea. I think I will try this out when I get a chance.