acmerobotics/road-runner-quickstart #28: How do I use a two wheel localizer with a Gyro?

EddieDL, 2019-10-26:
I attempted to Modify the StandardTrackingWheelLocalizer to use a Gyro for heading. After getting everything setup, it gives an error that "The specified configuration cannot support full localization". What am I doing wrong?

rbrott, 2019-10-26:
Can you provide the specific wheel positions and directions? This message means that your configuration doesn't have enough "variety" to localize in all three dimensions (x, y, heading).

EddieDL, 2019-10-26:
This might explain my issue. We were using a right and left wheel (both 0 degrees). From what you are saying we would need an x and y wheel (0 degree and 90 degree).   If we do that, does the track width matter?

rbrott, 2019-10-26:
Yes, you need separate x/y wheels. In this configuration, I'm not sure what track width refers to. In general, position matters on the dimension perpendicular to the wheel direction.

EddieDL, 2019-10-26:
I am wondering how I calculate the `TRACK_WIDTH` value in `DriveConstants`. Perhaps that doesn't matter?

EddieDL, 2019-10-26:
Ok. We got it working and for whatever reason we are having much better luck with the two wheel localizer than we did with the three wheel. The three wheel never seemed to hold heading, based on our conversation in #25 I am guessing it is something else we are doing wrong :(

Thanks again for all your help.

closed 2019-10-26 by EddieDL