acmerobotics/road-runner-quickstart #29: [Q] Motor encoder PID & velocity PID

Calvin-Xu, 2019-10-31:
This is a conceptual question: we are wondering the role of the drivetrain PID loop.

We understand that the process of tuning the drivetrain PID coefficients with `DriveVelocityPIDTuner` is matching the measured velocities of each wheel to the target velocity according to the motion profile. 

We read that the built-in velocity PID (`RUN_USING_ENCODER`) tries to match the motor shaft's rotational velocity to the input power given ("In the `DcMotor.RunMode.RUN_USING_ENCODER` mode, the controller uses an onboard velocity PID controller, and the power is instead a scaled velocity.").

Since the wheel velocities are themselves calculated from the encoder readings on the drive wheels, we are wondering why the internal PID cannot handle it, and we need a separate closed loop for the drivetrain (i.e., if the built-in velocity PID makes the motors turn at the right rotational velocity, what more is needed to be done?). Of course our past experience suggests that this is not true, but we just cannot think it through. Any explanation would be of great help to our team!

rbrott, 2019-10-31:
>Since the wheel velocities are themselves calculated from the encoder readings on the drive wheels, we are wondering why the internal PID cannot handle it, and we need a separate closed loop for the drivetrain (i.e., if the built-in velocity PID makes the motors turn at the right rotational velocity, what more needs to be done?). 

Your reasoning is correct. `DriveVelocityPIDTuner` tunes the built-in PID coefficients; there is no second velocity PID loop. The default coefficients can work OK for some robots, although it doesn't hurt to check. The more error you can eliminate in each step, the better.

Calvin-Xu, 2019-10-31:
Thank you! Do you know if there is a way to query the default PID coefficients, as a starting point?

rbrott, 2019-10-31:
The tuning routine queries them initially. Just open the dashboard *after you start the op mode*.

closed 2019-11-02 by Calvin-Xu