acmerobotics/road-runner-quickstart #32: Incorrect theoretical Kv for RUN_USING_ENCODERS

henopied, 2019-11-03:
The Kv should assume around 5000 rpm bare motor speed similar to the ExH. Perhaps the ExH assumes 25% lower max? Either way the current Kv doesn't get you up to speed correctly.

rbrott, 2019-11-03:
Have you found a discrepancy between the kV computed by Road Runner and the theoretical kV computed by hand? I'm interested in the specifics of your configuration.

Note that Road Runner leverages the motor definitions provided by the SDK. For some motors, the values in the SDK are not completely accurate (e.g., gear ratios may be listed as 20:1 instead of 19.2:1). This can be remedied by manually replacing the values in `DriveConstants` or writing your own custom configuration classes with the proper numbers.

henopied, 2019-11-03:
I configured it as a 5202 gobilda which has a 99.5 ratio. In DriveConstants, I set the ratio to 99.5/19.2 and it works great now. The max RPM roadrunner uses needs to be adjusted by achieveableMaxRPMFraction (default 0.85) from the configuration object when using velocity PID. I happened to guess a value that is almost exactly this and it worked well, but for different configurations this is a more robust solution. The achieveableMaxRPMFraction can be adjusted too which might be useful for this project.

closed 2019-11-06 by rbrott