acmerobotics/road-runner-quickstart #35: Negative heading error

Calvin-Xu, 2019-11-06:
The heading error seems to be the negative of what is expected. This happened to us while testing `TrackWidthTuner` and `FollowerPIDTuner`. The heading value seems correct, and the dashboard output is as followed:

[image]

Translational error seems to be correct. We feel that we definitely made a mistake in our configurations. Would appreciate any pointers!

[configurations.zip](https://github.com/acmerobotics/road-runner-quickstart/files/3812135/configurations.zip)

[image]

rbrott, 2019-11-06:
Could you elaborate on the issue specifically? I'm not sure what the setpoint is, so I can't really assess whether the error is correct or not.

Calvin-Xu, 2019-11-06:
So the following dashboard output happened while we were running `TrackWidthTuner`, which I believe sets the target heading to 180 degrees (pi rad), turns the robot, then clears the drive's PoseEstimate and repeats the process.

The heading values seem correct as the robot turns counterclockwise and stops at 180 degrees. One would expect that the error would initially be pi (rad), then steadily decrease to zero as the turn is completed. However, the error is initially zero and decreases as the robot is turning, ending up at -pi rad (when we expect it to be zero). We are not sure what could have caused this.

Calvin-Xu, 2019-11-07:
I hope this may be useful: heading and heading error when the turning angle is set to 270 degrees in `TrackWidthTuner` (window 10000ms).

[image]

rbrott, 2019-11-07:
Can you try upgrading RR to v0.4.6?

Calvin-Xu, 2019-11-07:
Thank you! We'll be sure to test again once we get back to it tomorrow.

Calvin-Xu, 2019-11-07:
The problem seems to persist; we have tried both `StandardTrackingWheelLocalizer` and `MecanumDrive.MecanumLocalizer`, and find the issue with either one of them; Still heading & all of odometry is correct, but heading error is not.

We are checking our drive configurations again. Do you think there is an issue with that / we can try some workaround?

Calvin-Xu, 2019-11-08:
So to debug / attempt a terrible workaround we tried adding pi to `getLastError()` in `SampleMecanumDriveBase`, as we think heading error seems to be shifted down by pi:

```java
public Pose2d getLastError() {
        switch (mode) {
            case TURN:
                return new Pose2d(0, 0, turnController.getLastError() + Math.PI);
        }
        throw new AssertionError();
    }
```

The output of the dashboard while running `TrackWidthTuner` is the following:

[image]

We find it interesting that pi seems to be subtracted from heading error at the start of each turn, probably being the source of the error, and trying to find out where that happened.

Calvin-Xu, 2019-11-08:
If this may be of help, we set breakpoints on `getError()` while running `TrackWidthTuner ` and took two screenshots:

[image]

[image]

commit 86454df2 (2019-11-09): Fixes #35 and #36
closed 2019-11-09 by rbrott