acmerobotics/road-runner-quickstart #36: Incorrect error calculation for `turnController`

Calvin-Xu, 2019-11-09:
We have been trying to debug with the following issue: [Unexpected heading error behavior](https://github.com/acmerobotics/road-runner-quickstart/issues/35), and have some speculations regarding its cause:

### If we understand correctly...

Since heading error seems to be correct when following a trajectory, we decided that it may be a problem only affecting the drive's TURN mode. It appears that in `SampleMecanumDriveBase` the `turnController` is separate from the follower:

```java
private PIDFController turnController;
. . .
follower = new HolonomicPIDVAFollower(TRANSLATIONAL_PID, TRANSLATIONAL_PID, HEADING_PID);
```

The follower has three three additional `PIDFController`s for translational and heading control:

```java
private val axialController = PIDFController(axialCoeffs)
private val lateralController = PIDFController(lateralCoeffs)
private val headingController = PIDFController(headingCoeffs)
```

Since the heading control in `FOLLOW_TRAJECTORY` seems to work, we are interested in how `headingController` in the follower is different from `turnController` in the drive. They are both `PIDFController`s, which seems to set the target position to zero by default.

```java
/**
* Target position (that is, the controller setpoint).
*/
 var targetPosition: Double = 0.0
```

In the `HolonomicPIDVAFollower`, error seems to set to the `PIDFController`s by setting setpoint = error (actually right in the comments)

```java
// you can pass the error directly to PIDFController by setting setpoint = error and position = 0
axialController.targetPosition = poseError.x
lateralController.targetPosition = poseError.y
headingController.targetPosition = poseError.heading
```
### What we speculate the error is
But it seems to us that error is not set to `turnController` in this manner. Its `targetPosition` seems to be never set to and therefore remains 0. Error calculations seem to be based on it. The debugger output seems to confirm this, as well as the behavior of the heading error (starting at 0, becoming increasingly negative as the counterclockwise turn is executed) we experienced in 
[Issue #35](https://github.com/acmerobotics/road-runner-quickstart/issues/35).

[image]

[image]

We only had a cursory look of the library, and cannot explain some unexpected behaviors we encountered with this explanation. Nevertheless we hope this may be of use in some way.

rbrott, 2019-11-09:
Thanks for identifying the cause. The solution is similar to the PR except the error is updated with the profile at each time step instead of once at the start of the turn.

closed 2019-11-09 by rbrott

Calvin-Xu, 2019-11-09:
Thank you for your support!

Calvin-Xu, 2019-11-10:
After more careful thinking we realized why our proposed solution is quite wrong and have a better understanding of motion planning & feedforwards. This has been an interesting exercise, and thank you for your support.