acmerobotics/road-runner-quickstart #39: [Q] StandardTrackingWheelLocalizer REV Bulk Read

Calvin-Xu, 2019-11-15:
We read in the comments to StandardTrackingWheelLocalizer that "Note: this could be optimized significantly with REV bulk reads", which we are trying to do.

We wonder if they best practice would be implementing a `getTrackingWheelPositions` in the drive class, then set the values to a `List<Double>` of the localizer, which is returned when the localizer's `getWheelPositions()` is called. If this is not the ideal way of doing it, we hope our question may be helpful to other teams.

rbrott, 2019-11-15:
No there's no need to add methods to the drive class. Just replace the logic in the localizer's `getWheelPositions()` with a call to get the bulk data and return the proper values. You can also compare optimized and non-optimized versions of the drive classes to get a better idea of the necessary changes.

Calvin-Xu, 2019-11-15:
Thanks. So is it advisable to do one bulk read per cycle of the main OpMode and cache the data for all functions that may require it (in drive, localizer & others)? Or does this have no advantages to calling for bulk data in each function that needs it?

rbrott, 2019-11-15:
Ok things are a bit more nuanced. My previous comment assumes that you have drive encoders on one hub and dead wheel encoders on another hub. If this is not the case, you will have to share the bulk reads between the two methods. In the RE2 interface, calling `getBulkInputData()` is guaranteed to trigger a command, so you don't want to call it additional times in your loop. I personally recommend some kind of bulk cache object that is instantiated once per op mode and is cleared/reset at the beginning of each loop. You can pass this object to all of your subsystems in the ctor, and they can retrieve that cached bulk data as much as they like.

closed 2019-12-01 by rbrott