acmerobotics/road-runner-quickstart #40: [Q] What is a sane MaxJerk range?

EddieDL, 2019-11-15:
I am wondering if there is some way to quantify a sane MaxJerk range. If you have maxVelocity=50 and maxAcceleration=20 what is a sane way to think about the MaxJerk value?  Can we consider it a relation to the maxAcceleration?

The reason I am asking is that I started with a very small value (.01) and have tried a few things without success. I just don't quite understand what makes sense.

rbrott, 2019-11-15:
First of all, max jerk is ignored for the default profiles. These profiles use a series of max velocity and acceleration samples at fixed time intervals to generate the profile (instead of constant constraints for the entire profile). Unfortunately, this method is not trivially generalizable to jerk constraints. If you would like to limit jerk, you can make simple profiles with constant constraints using `SimpleTrajectoryBuilder` and friends. Note, however, that the curvature of the path will distort the profile and result in potential constraint violations in terms of robot or wheel velocities. You can read more about the different profile generation methods [here](https://github.com/acmerobotics/road-runner/blob/7477ac6b1a5a8dc3ab2516a8cfdffd855a18c6a9/core/src/main/kotlin/com/acmerobotics/roadrunner/profile/MotionProfileGenerator.kt#L16) and [here](https://github.com/acmerobotics/road-runner/blob/7477ac6b1a5a8dc3ab2516a8cfdffd855a18c6a9/core/src/main/kotlin/com/acmerobotics/roadrunner/profile/MotionProfileGenerator.kt#L297).

That said, I would try a max jerk of around twice your max acceleration. While it may be difficult to choose max jerk by itself, you can determine it from max accel and the time taken to reach constant velocity (coasting).

EddieDL, 2019-11-15:
>First of all, max jerk is ignored for the default profiles

Do you mean when you set a `maxJerk=0` or are you saying to use Jerk you need to take some additional actions?

> These profiles use a series of max velocity and acceleration samples at fixed time intervals to generate the profile (instead of constant constraints for the entire profile).

I don't understand "acceleration samples at fixed time intervals" as it relates to "generate the profile". I thought I understood the profile generation to happen one time before the profile is executed, and therefore the robot isn't moving.

> instead of constant constraints for the entire profile

If I am understanding things correctly, when you aren't using a `maxJerk` constraint there is some sort of "automatic" jerk applied in the initial part of the motion. If you use `maxJerk` it is applied to the entire motion and is really intended to attempt to add a static "compensation"? My original assumption was, 'maxJerk` is intended to tune the start of the motion related to overcoming static friction.

Overall I guess I am trying to understand if it is common/important to apply `maxJerk` to a robot drivetrain to ensure accuracy? 

We are currently in a situation where we have very repeatable motion but it isn't accurate. Over long distances we are maybe 6-10 inches off and over short distances we are of 1-2 inches. To resolve this I wanted to "fix" our randomly selected `maxJerk=.01`. 

It sounds like our current `maxJerk` value is ridiculously low (which I kind of thought). Now we need to decide if we should just set `maxJerk=0` or try to actually tune that value.

Side note for anyone who is wondering if this question is answered in the docs. While I was writing this I did find a decent **[example](https://acme-robotics.gitbook.io/road-runner/tour/motion-profiling)** in the docs. I had read this part of the docs before, it just didn't jump out at me what was going on. 

[image]

[image]

rbrott, 2019-11-15:
> > First of all, max jerk is ignored for the default profiles
> 
> Do you mean when you set a `maxJerk=0` or are you saying to use Jerk you need to take some additional actions?

For default profiles, the value of max jerk does not matter. It will be ignored regardless. For simple profiles, the value does matter. In that case, a max jerk of 0 generates an acceleration-limited profile.

> I don't understand "acceleration samples at fixed time intervals" as it relates to "generate the profile". I thought I understood the profile generation to happen one time before the profile is [...]

Yes, that is true. The samples I mentioned are displacements along the path. This is essential to map the displacement velocity/acceleration to match the local geometry of the path (e.g., the curvature). These samples are collected offline after the path has been constructed and are used to plan the trajectory.

> If I am understanding things correctly, when you aren't using a `maxJerk` constraint there is some sort of "automatic" jerk applied in the initial part of the motion. If you use `maxJerk` it is [...]

Jerk has nothing to do with static friction. For our purposes, the only relevant static friction is inside the motors. This is compensated by one of two mechanisms:
* Motor velocity PID control
* kStatic constant velocity/power feedforward

> Overall I guess I am trying to understand if it is common/important to apply `maxJerk` to a robot drivetrain to ensure accuracy?

Max jerk should not have a major impact on the accuracy of your motions. Simple/constant constraint profiles are not commonly used for drivetrains.

> We are currently in a situation where we have very repeatable motion but it isn't accurate. Over long distances we are maybe 6-10 inches off and over short distances we are of 1-2 inches. To resolve this I wanted to "fix" our randomly selected `maxJerk=.01`.

I can guarantee you that max jerk is not the culprit here. It sounds like there may be something off about your kV. If you're using the theoretical value, double check the motor configuration; there are occasionally slight errors in the SDK definitions.

EddieDL, 2019-11-15:
> For default profiles, the value of max jerk does not matter. It will be ignored regardless. For simple profiles, the value does matter. In that case, a max jerk of 0 generates an [...]

How do I know if I am using default or simple profiles? I am using  `TrajectoryBuilder.lineTo()` primarily but I have also used `TrajectoryBuilder.back()` etc. Do I just have to go into the details of those commands to determine what they do or perhaps all of them are "default"?

> Jerk has nothing to do with static friction. For our purposes, the only relevant static friction is inside the motors. This is compensated by one of two mechanisms:
>
> Motor velocity PID control [...]

My familiarity with the term Jerk is limited. I thought Jerk is what gives you the slopes on the `a(t)` line in the following graph. The purpose being to account for the fact that you can't just go from a velocity of `0` to `X` instantaneously. It takes time for the physical components to actually change, due to some other force.
[image]

If you could verify my new understanding I would appreciate it. Generally speaking, you should not configure a `maxJerk` unless you are dealing with a simple/constant force being applied to the mechanism (as in your example, gravity on an elevator). Instead utilize the default profiles as they will dynamically adjust to the desired demand without the need for a Jerk setting.

> I can guarantee you that max jerk is not the culprit here. It sounds like there may be something off about your kV. If you're using the theoretical value, double check the motor configuration; there [...]

In our case we have measured our max rpm's and the `kV` is set to the default of `1 / rpmToVelocity(getMaxRpm()`. This is because we are now using the Rev Velocity PID control. With that said, I am not super comfortable that the PID values for that are tuned very well. We struggled to figure out how to precisely tune them. I am not sure if you have any suggestions about exactly how to tune this, but we simply ran the `DriveVelocityPIDTuner` while graphing the errors and attempted to make changes that would smooth the error out. In the end we just left the "default" PID values from Rev because we couldn't identify better values. 

One note/question: I assume the Rev PID values aren't stored over power reset.

Sorry this is getting long, but this is really helping me understand this issue. I appreciate it.

rbrott, 2019-11-15:
> How do I know if I am using default or simple profiles? I am using `TrajectoryBuilder.lineTo()` primarily but I have also used `TrajectoryBuilder.back()` etc. Do I just have to go into the details of those commands to determine what they do or perhaps all of them are "default"?

`TrajectoryBuilder` always uses default profiles; `SimpleTrajectoryBuilder` always uses simple profiles.

> My familiarity with the term Jerk is limited. I thought Jerk is what gives you the slopes on the `a(t)` line in the following graph. The purpose being to account for the fact that you can't just go [...]

Yes, jerk is the derivative of acceleration. Jerk-limited profiles prevent instantaneous acceleration transitions (i.e., asymptotic jerk). Qualitatively it smooths the profiles further to ease wear on mechanisms and avoid exciting higher frequency vibrational modes. If you like, limiting jerk prevents the instantaneous/sudden application of force.

> If you could verify my new understanding I would appreciate it. Generally speaking, you should not configure a `maxJerk` unless you are dealing with a simple/constant force being applied to the [...]

The use of jerk-limited profiles is independent of the underlying plant. It serves no feedforward purpose (there is no kJ); classical models stick to forces and largely ignore higher order effects. You can use jerk-limited profiles on linear slides regardless of whether they're fighting gravity. The "default"/complex profiles have a specific place in the 2D paths. In this case, there are two coupled degrees of freedom whose relationship depends on the geometry of the path. Therefore it is necessary to consider this geometry when constructing the profiles; this method cannot handle jerk constraints.

> In our case we have measured our max rpm's and the `kV` is set to the default of `1 / rpmToVelocity(getMaxRpm()`. This is because we are now using the Rev Velocity PID control. With that said, I am [...]

First of all, there are new changes that adjust the theoretical kV a little in the case of motor velocity PID. To provide headroom for the PID algorithm, the provided power values are scaled down so the power range [-1, 1] now corresponds to 0.85*[-max_vel, max_vel]. This factor of 85% was neglected until recently. It sounds like this will fix your accuracy issues.

As for tuning, you want to match the measured velocity to the target velocity as best as possible. It's most important to eliminate phase lag (translational separation between the graphs). From there, try to eliminate oscillations as best as you can (although it's fine if some small oscillations remain).

> One note/question: I assume the Rev PID values aren't stored over power reset.

Correct. If you store the values in `DriveConstants`, the provided drive classes will set them at the beginning of each op mode.

EddieDL, 2019-11-15:
This all makes a lot more sense now, I was just missing a few pieces.

> First of all, there are new changes that adjust the theoretical kV a little in the case of motor velocity PID. To provide headroom for the PID algorithm, the provided power values are scaled down so [...]

Are you talking about changes to this repo, specifically [this](https://github.com/acmerobotics/road-runner-quickstart/commit/2c4968515de06f094afc0d51fb3c564c3ab32b3d) changeset? I also noticed there is a newer version of roadrunner as well.

rbrott, 2019-11-15:
Yes, I'm talking about [this commit](https://github.com/acmerobotics/road-runner-quickstart/commit/7ecb2fe6209b80e5348e4d0de0cd07b36daa73d8#diff-49440fa8b02e28776e39c2de90ac85f1R85). There is a newer version of road runner core to fix a major bug in the localizers.

EddieDL, 2019-11-15:
I just went through all of the updates (you guys have been busy and I should have been following this repo 🥇). I think I understand most of the changes and see how it will likely make MANY of the issues we have been having better. 

One question I have is in regard to how to get max rpm. If I have already empirically gathered a value, is it worth still using that? If it won't provide a significantly better operation I would prefer what you have now.

rbrott, 2019-11-15:
If you're using the motor velocity PID, I would stick with the theoretical kV. If there's still an accuracy problem, double check all of the drive constants.

EddieDL, 2019-11-17:
When you are running DriveVelocityPIDTuner how fast should the robot be going? Right now when I am testing it maxes out around 2 inches per second, which seems incredibly slow and not reflective of how we would be running the robot. I am trying to figure out if I have something else wrong or if that is how that is supposed to work.

closed 2020-03-14 by rbrott