acmerobotics/road-runner-quickstart #42: [Q] Using setDrivePower in TeleOp

Calvin-Xu, 2019-11-22:
So we see that `setDrivePower` is used in `LocalizationTest` that enables gamepad controlled drive:

```java
drive.setDrivePower(new Pose2d(
                    -gamepad1.left_stick_y,
                    -gamepad1.left_stick_x,
                    -gamepad1.right_stick_x
            ));

            drive.update();
```
We noticed that the translational and angular velocities cannot seem to be controlled simultaneously. Would this be the expected behavior due to some physical constraints?

rbrott, 2019-11-22:
Yes, there are physical constraints that are neglected. This is commonly dealt with by "normalizing" the wheel powers to a max of 1.

Calvin-Xu, 2019-11-22:
Thank you. Would you elaborate where the "normalizing" process occurs? and is it the same as a cap on actuator speed?

Additionally, we wonder what steps might we take to enable both translational and angular velocity control, or would it not be realistic?

rbrott, 2019-11-22:
The normalizing process is not a part of Road Runner. It is deliberately excluded to make teams think about limiting the velocities before passing them on to RR. Nevertheless, if you would like, you can normalize the powers in `setMotorPowers()` by dividing all powers by `max(1, |max_power|)`.

Looking back at this code, I realized it's a little inconsistent when it comes to angular velocity; `omega` is treated as having real units (`rad/sec`) instead of some fraction of max omega. Taking that into consideration, `v_x`, `v_y`, and `omega` (`v_x` and `v_y` are powers) must satisfy the following relation to stay within the domain of achievable wheel velocities:
`|v_x| + |v_y| + |omega * (track_width + wheel_base) / 2| <= 1`
Simultaneous translation and rotation should be perfectly achievable if this constraint is obeyed.

rbrott, 2019-12-01:
I just added some normalization code to `LocalizationTest` if you want to take a look. It allows you to naturally weight the configuration dimensions (e.g., weight omega to prefer turning over translation if the constraints are violated).

EddieDL, 2019-12-02:
Note if you are trying to get the changes @rbrott committed to work you need to make sure you update to the latest version of Road-Runner (0.4.7+). There were some changes to Road Runner to address the second part of his last comment. The new version respects `omega` being a fraction of max omega.

> Looking back at this code, I realized it's a little inconsistent when it comes to angular velocity; omega is treated as having real units (rad/sec) instead of some fraction of max omega. Taking that [...]

Thanks to @rbrott for changing this I was just going to investigate this further :)

Calvin-Xu, 2019-12-11:
Thank you so much for the example implementation!

closed 2019-12-11 by Calvin-Xu