acmerobotics/road-runner-quickstart #43: [Q] How does Max Velocity work when you are moving in two directions at once?

EddieDL, 2019-12-02:
We have our robot tuned in fairly well. Our only remaining issue is when we add motions in two directions. Doing this create inconsistent profiles. I believe this is because there is some assumption that we can move max velocity in both directions at the same time, which we can't. Is this a valid concern?

```
Trajectory complexMove = drive.trajectoryBuilder()
                .lineTo(new Vector2d(40, 19), new LinearInterpolator(0, 0))
                .lineTo(new Vector2d(80.5, 32), new LinearInterpolator(0, 0))
                .build();
        drive.followTrajectorySync(complexMove);
```

My expectation is that this will create a "triangle" motion. Which it does, but we never really reach the correct Y. This issue becomes worse the more we increase our max velocity. It seems like the only solution is to greatly reduce the max velocity significantly.

Should we be thinking about this differently?

rbrott, 2019-12-03:
No, it should not create inconsistent profiles. If you're using `DriveConstraints` directly, the configuration velocity magnitude is limited (this can present issues if you turn too much). If you're using drivetrain-specific constraints (e.g., `MecanumConstraints`), the wheel velocities are limited.

Is the localizer aware of the robot's position or does it drift as well?

EddieDL, 2019-12-03:
We are using `DriveConstraints`. I am not exactly sure how I might use a `MechanumConstraints`, but that seems more like what we would want.

I haven't verified on the specific path I provided above, but in general the localizer is aware the position is not correct (in the dashboard the dot for the robot doesn't match the line that is drawn).

rbrott, 2019-12-03:
If you're using the quickstart, you're probably using `MecanumConstraints` instead of `DriveConstraints`. The former is constructed with an instance of the latter.

If the localizer is tracking the position, it's just a matter of tuning the controllers. Don't be afraid to try more aggressive tunings.

EddieDL, 2019-12-03:
We are using a "customized" version of the quickstart, but almost identical.

I am confused by where `MechanumConstraints` is used in the quickstart. Below is what I see in `SampleMecanumDriveREVOptimized.java`. I see similar in `SampleMecanumDriveBase.java`. 
 
[image]

Also, is it possible to tune just in one direction (X vs Y)?

rbrott, 2019-12-03:
Yes, the follower accepts separate coefficients for X and Y. The quickstart passes the same translational coefficients to both, but you that can easily be changed.

You could also modify the feedforward to add a multiplier for lateral movement, but that requires modifying and redistributing the core library.

EddieDL, 2019-12-03:
Sorry to be so dense. I was completely confusing `DriveConstraints` with `DriveConstants`. 

The line right below what you posted is where you set the PID for X vs. Y.

https://github.com/acmerobotics/road-runner-quickstart/blob/5f038638a1d35dcc69957b2e25b5d0795b28ee2c/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/mecanum/SampleMecanumDriveBase.java#L79

closed 2019-12-03 by EddieDL