acmerobotics/road-runner-quickstart #44: How to use translational and heading pid?

LetianLi, 2019-12-10:
It is not clear on how to use FollowerPIDTuner to tune the TRANSLATIONAL_PID and HEADING_PID found inside of MecanumDriveBase. It is also not clear on what these PID variables do and how it effects the movement of the robot. Help would be appreciated. Thanks

rbrott, 2019-12-10:
The followers close the loop on the robot's position relative to the path/trajectory. Tuning the follower PID loops ensures that the robot actively tracks the paths/trajectories instead of relying entirely on feedforward. When you run the tuner, the path the robot traces will inevitably drift without feedback. Your job is to correct for this divergence using the PID gains (start with P; you should be able to eliminate most of the error without I or D).

LetianLi, 2019-12-10:
Specifically, what do each pid correct in how the robot moves?

EddieDL, 2019-12-10:
The TRANSLATIONAL_PID corrects movement in the X and Y direction. The HEADING_PID corrects the rotation.

LetianLi, 2019-12-10:
How would these pids correct in multiple directions. The robot's heading can be drifting left or right. I believe PID can only go up.

EddieDL, 2019-12-10:
You should probably do some reading [here](https://acme-robotics.gitbook.io/road-runner/tour/pid-control). It provides some details about PID.

These values are used to adjust the output to the motors and keep the position or heading on the provided target.  Each one is used as a component of the final output and can be negative or positive.

LetianLi, 2019-12-11:
I am assuming heading pid will correct drifting (for example, there are different weights on each mecanum wheel), and translational pid will correct how far to move (if you are short of trajectory end).
Is this correct?

EddieDL, 2019-12-11:
HEADING_PID corrects the rotation of the robot and the TRANSLATIONAL_PID corrects the position. 

For example if you attempt to move the robot "forward" 10 inches with no change in heading, the HEADING_PID will attempt to ensure the robot does not change heading while the TRANSLATIONAL_PID makes sure the robot reaches 10 inches properly.

Also, you should not be thinking about these PID values applying to the overall destination. Road-Runner will create a Motion Profile ([more reading](https://acme-robotics.gitbook.io/road-runner/tour/motion-profiling)). This means it is not just trying to get to the end target. It is using the other constraints like max velocity, max acceleration, etc to plan out what the position should be at any given point in the motion. 

This means the PID is not saying my target is 10 inches away, how fast should I be going. It actually slices that movement up into small points in time. These have expected position and headings for each segment. Each time you update the motor output using these PID values, they are using the target from the segment, not the end target.

This is how you can create complex motions without having to stop in between. To Road-Runner it is just more points in the trajectory that it will attempt to follow.

@rbrott feel free to correct me, I still feel like a novice at all of this :)

closed 2019-12-24 by LetianLi