acmerobotics/road-runner-quickstart #46: Tuning motor PIDF values

jasonsue53, 2019-12-20:
I know this is not exactly a question about road runner but it seems like a basic we need to fully understand and set correctly before properly start tuning road runner.  We have a mecanum chassis with 40:1 Rev Hex motors.  

1.  The FIRST Global instructions for determining max velocity doesn't specify whether that test should be done with the wheels free spinning or if the wheels should be on the ground propelling the mass of the robot.  
2. For now we assumed the test should be done with the wheels on the ground propelling the robot.  When we ran the test we got slightly different max velocities for each motor (between 3200 and 3360) and have calculated the PIDF for each motor.  However, in the road runner quick start the same PID coefficients are applied to all motors.  Should we change this code so we can apply unique coefficients for each motor or is the differences too small to make a difference? 
3. Trying to understand why F is calculated the way it is in DriveConstants rather than being a value to set along with the other coefficients.  It seems to want to use the theoretical F for the rev hex motors rather than what we found empirically through our max velocity test.

rbrott, 2019-12-20:
> Trying to understand why F is calculated the way it is in `DriveConstants` rather than being a value to set along with the other coefficients. It seems to want to use the theoretical F for the rev hex motors rather than what we found empirically through our max velocity test.

You can use the empirical value if you would like (I'm curious if it's much different from the theoretical). At this point I'm not sure the marginal benefit is worth adding a step to the tuning procedure (also a maximum velocity tuner sounds like an easy way to break robots). The remaining slack can be addressed by the velocity PID.

jasonsue53, 2019-12-20:
Got it, thank you Ryan.  Using the properties of RevRobotics40HdHexMotor it calculates an F value of 11.7, based on a theoretical max velocity of 2800 ticks/sec.  When I did the max velocity test on our robot I got max velocities in the 3200 to 3360 range, so an F of 9.75 - 10.23.  You would have a much better idea of whether that difference is meaningful.   

Going through this whole process has given me a much better understanding of Drive Velocity PID tuning step so I'm going to be starting our tuning process over tonight by rolling back to that step.  You're a rock star, thanks again for everything you've done here!

closed 2019-12-20 by jasonsue53