acmerobotics/road-runner-quickstart #47: Very basic tuning question

jasonsue53, 2019-12-20:
Seems silly but I have to ask...  when running each test in the tuning process for the road runner quickstart it wasn't clear to me if we should be focused on the telemetry in the dashboard or the physical results.  For instance, in "StraightTest" should we be looking at what the telemetry is telling us about how close the robot landed to the target or should we be measuring with a tape measure how far the robot traveled and comparing with the target?

rbrott, 2019-12-20:
Both. The dashboard shows the robot's estimate of its position which should more-or-less match up with reality (some slight deviations are an inevitable consequence of dead reckoning). In the earlier tests, it's important to make sure the robot parameters have been set properly. These parameters are suspect whenever there is a systematic error in the robot's localization/motion. The dashboard becomes more important in the later stages when it becomes more difficult/tedious to assess the robot's tracking performance.

jasonsue53, 2019-12-20:
Quick follow on question.  Have you guys found a need to correct for those deviations over time through other methods like using lidar to measure distance from walls?  I'm assuming the deviations are additive over time and thus after lots of movement the deviation could become big enough to cause trouble for precise movement.

rbrott, 2019-12-20:
Yes, errors accumulate with any dead reckoning system. This accumulation with drive encoders is usually manageable during the auto period if the constraints are conservative enough and there are no collisions. If you want to go faster or experience general accuracy issues, you can try dead/tracking omni wheels as popularized by 11115. These systems are still based on dead reckoning but generally experience less drift and are more resilient to collisions/bumps. There are no popular absolute localization mechanisms for complete pose estimation. Some teams have had success with ultrasonic/ToF sensors, but this usually requires that the robot be at rest and it only corrects one dimension.

closed 2019-12-26 by jasonsue53