acmerobotics/road-runner-quickstart #48: Three Tracking Wheel Localizing

LetianLi, 2019-12-24:
When putting in the locations of the three tracking (dead) wheels, do we put in lateral distance of the outside, middle, or inside of the left and right wheels.
**I**||---------||**I** (outside distance), 
|**I**|---------|**I**| (middle distance), 
||**I**---------**I**|| (inside distance),

rbrott, 2019-12-24:
The middle distance. Ideally there is a single point of contact when the roller hits the floor.

LetianLi, 2019-12-24:
We have two points of contact as we have two omni wheels next to each other. Will this be a problem.
Our heading is currently 10 degrees off. We have all measurements that are relative to the center of the 4 mecanum wheels. How do we fix this error, all our measurements seem correct, including wheel radius.

LetianLi, 2019-12-24:
Our heading is currently 10 degrees off. We have all measurements that are relative to the center of the 4 mecanum wheels. How do we fix this error? all our measurements seem correct, including wheel radius.

rbrott, 2019-12-24:
Where are the other wheels? Are they all doubled?

LetianLi, 2019-12-24:
All of our wheels are doubled.
  new Pose2d(1 + 3/16, 6 + 13/16, 0), // left
  new Pose2d(1 + 3/16, -6 - 13/16, 0), // right
  new Pose2d(4 + 9/16, -10/16, Math.toRadians(90)) // front

rbrott, 2019-12-24:
OK `3/16` evaluates to `0`. You probably want `3.0/16.0` instead.

LetianLi, 2019-12-24:
Thank you! That solved the problem. I should have observed this stupid mistake myself!

closed 2019-12-24 by LetianLi

LetianLi, 2019-12-24:
After fixing the code, now the odometry wheel heading is about 0.2 to 0.5 degree difference compared to IMU, when turning 180 degree. Is that normal/acceptable? I think IMU (from rev hub) itself may have some errors too?

rbrott, 2019-12-24:
That seems acceptable. I'd compare after a bigger turn (i.e., a few revolutions) and see if the error is still acceptable.