acmerobotics/road-runner-quickstart #50: Turn tuning

jasonsue53, 2019-12-26:
We’re struggling with tuning our robot to predictably turn when doing a turn sync. 

1. We found that the empirical track width is different depending on how we have the track width set when we do the test. 
2. When doing the turn test we can tune the variables to get the turn to finish with a degree of the expected angle (ie 180 degrees) but when we test with a different angle it will be as much as 20 degrees off. 
3. What is the margin of error we should be able to achieve?  In other words, should we expect to be able to get within 2-3 degrees of the specified angle?

rbrott, 2019-12-27:
It sounds like there is something up with the drive constants/parameters. Have you tried running the localization test with your localizer? (And also with the default localizer if you're using tracking wheels.) You should be able to achieve a margin of error of at least a single degree if everything is tuned properly.

jasonsue53, 2019-12-27:
How do I run a localizer test?  I haven’t seen any mention of that in the documentation, sorry if I have missed it.  We’re using a Tile Runner chassis with 40:1 Rev HD Hex motors so we’re using the “..RevOptimized” class and the built in localizer.

jasonsue53, 2019-12-28:
Not sure if it's good news or bad but I just ran the localizer test and it worked beautifully.  Ran the bot all around the field and made a number of turns, strafes, etc. and then returned back to starting point the x, y, and heading were all within a couple inches/degrees.  I'd appreciate any other suggestions.

rbrott, 2020-01-01:
How is the velocity PID response? I assume there is no positional feedback at this point.

closed 2020-03-14 by rbrott