acmerobotics/road-runner-quickstart #51: How to effectively tune heading pid

LetianLi, 2019-12-31:
We have read the [Pid-Tuning](https://acme-robotics.gitbook.io/road-runner/tour/pid-control) document already. 
Heading PID is being really annoying to get right. We are just strafing left and right 60 in to tune.
This image is our P:5, I:0, D:0.03
We find that if I or D are greater than 0.05, the robot shakes a lot near the end of the strafe.
Error appears to stay up and then drop back down
[image]

The switch from positive to negative and vice versa is in every combination we have tested so far and is always near the end of the trajectory.
The current Heading PID is a little better in straight test (2 to -2 deg) but more spiky.

rbrott, 2020-01-01:
Does the issue go away at lower speeds? What are your constraints?

LetianLi, 2020-01-01:
Max velocity 45.0, Max acceleration 40.0, max jerk 0.0.

rbrott, 2020-01-02:
Does the issue persist if you lower the max accel to 25in/s^2 or similar? Also could you record the response after the trajectory ends? The follower will continue to correct if you pass it a non-zero `timeout` (see the docs).

LetianLi, 2020-01-03:
I can't find anything about timeout, nor the timeout in the code. The current picture has the start and end; after trajectory, errors default back to 0.

rbrott, 2020-01-03:
The `TrajectoryFollower` `timeout` parameter is not featured in the quickstart. It's mentioned in [kdoc](https://acmerobotics.github.io/road-runner/core/0.4.6/kdoc/com.acmerobotics.roadrunner.followers/-trajectory-follower/index.html)/javadoc. This will artificially extend the trajectory and continue correcting residual error.

LetianLi, 2020-01-04:
We end the trajectory fine, the problem is in the middle of our trajectory. X and Y also have a lot of error in the middle of trajectory but fixes at the end of trajectory.

rbrott, 2020-01-04:
It's probably just lag in velocity PID tuning. Being more aggressive there will be better than further position tuning. Some lag is somewhat inevitable, but I've found people generally back off too soon (i.e., the moment oscillation starts). Is there a problem with slightly larger transient error?

LetianLi, 2020-01-04:
Quite often when tuning velo pid, we see very large transient error when we begin decelerating. Sometimes it's just on one wheel, but a few other times, it's on most of them.

closed 2020-03-14 by rbrott