acmerobotics/road-runner-quickstart #54: Issues with trajectory yaml file, robot doesn't run

dvpatel, 2020-01-09:
Folks,

I'm new to RoadRunner, still learning the various APIs and integration model.  One of the test I was trying to run was to build a simple trajectory (go straight for 48") using the GUI plugin and run it as autonomous.  The YAML file seems to load fine, I think, but the robot doesn't go anywhere.  Any suggestions on how best to diagnose and resolve this issue?  Thank you!  By the way, love the toolkit.

Sample code:
@Autonomous(group = "drive")
public class TrajectoryTest extends LinearOpMode {

    private static final String TRAJECTORY_FILE = "MyTest";

    @Override
    public void runOpMode() throws InterruptedException {
        SampleMecanumDriveBase drive = new SampleMecanumDriveREV(hardwareMap);
        Trajectory traj = null;
        try {
            traj = AssetsTrajectoryLoader.load(TRAJECTORY_FILE);
        } catch (IOException e) {
            throw new InterruptedException("Trajectory Error.");
        }
        waitForStart();

        if (isStopRequested()) return;
        if (traj != null) {
            drive.followTrajectorySync(traj);
        }
    }
}

Trajectory YAML:
---
startPose:
  x: 0.0
  y: 0.0
  heading: 0.0
startHeading: 0.0
steps:
- pose:
    x: 48.0
    y: 0.0
    heading: 0.0
  interpolationType: "LINEAR"
  heading: 3.141592653589793
resolution: 0.25
version: 2

rbrott, 2020-01-09:
It looks like you're using the new v2 trajectory file format which is still in the works and not fully supported by the quickstart.

dvpatel, 2020-01-09:
Thanks for the response.  I'm assuming v1 trajectory file may work.  Any examples / docs for v1 format?

rbrott, 2020-01-09:
No, there aren't any docs. It's mostly intended for use with the GUIs. That said, it's a serialized version of [this class](https://github.com/acmerobotics/road-runner/blob/master/core/src/main/kotlin/com/acmerobotics/roadrunner/trajectory/config/LegacyTrajectoryConfig.kt).

dvpatel, 2020-01-21:
Thanks for your help.

closed 2020-01-21 by dvpatel