acmerobotics/road-runner-quickstart #55: Localization Method

Mihneaghitu, 2020-01-17:
I cannot seem to find a proper localization method for TrackWidthtTuner. I am using encoders, I think I did VelocityPIDTuner and StraightTest ok and in the code it says to set a localization method that does not rely on encoders for determining position.As I saw StandardTrackingWheelLocalizer relies on encoders and if I use a basic Localizer, I do not know how to implement the three methods (getPoseEstimate, setPoseEstimate and update). Any help would be extremely much appreciated.

rbrott, 2020-01-17:
The default localizer and the tracking wheel localizer both do **not** use drive encoders for heading computation.

Mihneaghitu, 2020-01-17:
Do I need to change something in the StandardTrackingWheelLocalizer or is it ok as it is (because my configuration is 4 mecanum drive wheels, not as it shows there) ?

rbrott, 2020-01-17:
You should not use `StandardTrackingWheelLocalizer` if you do not have tracking wheels. The default localizer uses the drive encoders which appears to be what you need (i.e., no call to `setLocalizer()` is necessary).

Mihneaghitu, 2020-01-17:
So the setLocalizer() method doesn't need to appear in the mecanum drive hardware config or in the TrackWidthTuner class if using the default Localizer?

rbrott, 2020-01-17:
No

Mihneaghitu, 2020-01-17:
Thank you so much for the responses, they helped a lot!

closed 2020-03-14 by rbrott