acmerobotics/road-runner-quickstart #56: lag upon initialization

LetianLi, 2020-01-24:
Our phones have 2+ seconds of initialization normally and when the imu problem message pops up, 5+ seconds. This is a big problem for competition and the switch from auto to teleop.

rbrott, 2020-01-24:
This doesn't seem like a Road Runner issue. Perhaps you should file an issue [here](https://github.com/FIRST-Tech-Challenge/SkyStone).

LetianLi, 2020-01-24:
the lag happens when initializing with road runner. That seems to be the main problem.

rbrott, 2020-01-24:
If you don't use the IMU, there's no need to initialize it. In this respect, the quickstart code is just like a normal op mode.

LetianLi, 2020-01-24:
Even without the imu, it takes a while.

LetianLi, 2020-01-24:
would calling the road runner libraries cause lag?

rbrott, 2020-01-24:
Probably not, but it's possible with misuse (e.g., arc length parametrization of a spline with high curvature). You should profile and find out for yourself.

closed 2020-03-14 by rbrott