acmerobotics/road-runner-quickstart #57: Using followTrajectory (non synchronous)

Calvin-Xu, 2020-01-28:
We are trying to use the `followTrajectory` API, and created the following minimal OpMode to test it:

```java
@TeleOp
public class MinimalFollowTrajectoryDemo extends LinearOpMode {

    private SampleMecanumDriveREVOptimized drive;
    private ExecutorService driveUpdateExecutor;
    private Runnable driveUpdateRunnable = () -> {
        while (!Thread.currentThread().isInterrupted()) {
            drive.update();
        }
    };

    @Override
    public void runOpMode() throws InterruptedException {
        driveUpdateExecutor = ThreadPool.newSingleThreadExecutor("drive update");
        drive = new SampleMecanumDriveREVOptimized(hardwareMap);

        waitForStart();

        driveUpdateExecutor.submit(driveUpdateRunnable);
        Trajectory testTrajectory = drive.trajectoryBuilder()
                .forward(20)
                .build();

       drive.followTrajectory(testTrajectory);
    }
}
```

We don't want to block the main thread while following a trajectory, so that we can control other systems while following a trajectory. However, in our test OpMode the drive does not run the desired distance, and the motion is extremely choppy. Changing `followTrajectory ` to `followTrajectorySync` displays the behavior we desire.

We are not sure what we are implementing wrong compared with the `waitForIdle()` in `followTrajectorySync`. It seems to us that the drive is not updated frequently enough. Any help is appreciated!

rbrott, 2020-01-28:
Yes, your assessment that the drive is not updated frequently enough is likely correct. The use of threading in this instance misses the point of the asynchronous API. Calling `followTrajectory()` doesn't block the main thread and allows you to handle other asynchronous tasks concurrently with interleaved updates. As the number of concurrent tasks and distinct "phases" grows, it becomes advantageous to use a state machine.

Here's a contrived example that demonstrates these principles:
```java
drive = ...;
distanceSensor = ...;
armServo = ...;

waitForStart();

drive.followTrajectory(...);
while (opModeIsActive() && drive.isBusy()) {
    if (distanceSensor.getDistance() > 5) {
        armServo.setPosition(0.37);
    } else {
        armServo.setPosition(0.78);
    }
    drive.update();
}
```

This approach gives more predictable performance than threads. Only one command can be transmitted and received at a time; therefore, there is no benefit to multiple hardware threads. Furthermore, by partitioning the control loop without proper synchronization, you lose control over execution frequency and sequencing.

Calvin-Xu, 2020-01-28:
Thank you! Would you provide an example of following a series of trajectories under this paradigm?

rbrott, 2020-01-28:
Simply replicate the sample.

```java
drive.followTrajectory(firstTrajectory);
while (opModeIsActive() && drive.isBusy()) {
    // ...
    drive.update();
}

drive.followTrajectory(secondTrajectory);
while (opModeIsActive() && drive.isBusy()) {
    // ...
    drive.update();
}
```

Calvin-Xu, 2020-01-28:
So after having used quickstart for a while, we are thinking of adopting 1. stateful hardware subsystems 2. asynchronous update API for subsystems to achieve the desired states.

We actually do not plan to use multiple hardware threads, but only one that updates all hardware subsystems. What we have in mind is more akin to this:

```java
drive = ...;
elevator = ...;
hardwareUpdateExecutor = ...;

private Runnable hardwareUpdateRunnable = () -> {
	while (!Thread.currentThread().isInterrupted()) {
		drive.update();
		elevator.update();
		}
	};

hardwareUpdateExecutor.submit(hardwareUpdateRunnable);
	
waitForStart();

drive.followTrajectory(...);
elevator.toPosition(...);
```
Which seems to eliminate the need to manually control when to call the various `update()` in the OpMode control logic. Is there something that might dissuade us doing so (apart from that the drive did not behave as expected in our testing)?

rbrott, 2020-01-29:
No, there isn't necessarily an issue with that besides the normal prudence when using threads.

Calvin-Xu, 2020-01-29:
Thanks, but we seem to run into issues with this approach (see initial question). It seems the update loop on the hardware thread is not running fast enough? We will try to log it the update times, but we are not sure what might be causing the problem in a minimal OpMode like this.

rbrott, 2020-01-29:
> However, in our test OpMode the drive does not run the desired distance, and the motion is extremely choppy.

I suppose this merits elaboration. Is it stopping prematurely? Are you blocking the main thread long enough for the trajectory to finish executing? In your case, you want a version of `followTrajectorySync()` that doesn't call `update()`. Are you stopping the thread when the op mode terminates?

Calvin-Xu, 2020-01-29:
> Is it stopping prematurely?

I have not timed it, but I think it ran for the same time duration. The motion was jagged and choppy—I supposed that motor powers are not being set frequently enough.

> Are you blocking the main thread long enough for the trajectory to finish executing?

Do we need to block the main thread? I supposed that after `followTrajectory` is set the OpMode would not exit until the trajectory has been followed.

> In your case, you want a version of followTrajectorySync() that doesn't call update().

I am not sure if I get that. Is not `followTrajectorySync() ` just `followTrajectory` and `waitForIdle()`, which keeps invoking `update()` until the trajectory is finished.

> Are you stopping the thread when the op mode terminates?

No. So that was the entire OpMode. Should we shut down the executor in the end?

rbrott, 2020-01-29:
> Do we need to block the main thread? I supposed that after `followTrajectory` is set the OpMode would not exit until the trajectory has been followed.

Yes, you need to block the main thread in a linear op mode. When `runOpMode()` returns, `requestOpModeStop()` is called.

> Should we shut down the executor in the end?

Yes, call `executor.shutdownNow()`.

Calvin-Xu, 2020-01-29:
Thank you, we will test that tomorrow. On a side note, are there any recommended resources to learn the life cycle of OpModes and how they are run?

rbrott, 2020-01-29:
Not really. The best resource is probably the source itself.

closed 2020-02-05 by Calvin-Xu