acmerobotics/road-runner-quickstart #59: FTC Dashboard

mihir-jain, 2020-02-09:
When I try to run FTC Dashboard, the side of the screen, starts to blank out, and the most common "mode" is saying that it isn't active

rbrott, 2020-02-09:
You'll have to elaborate. Do you have a screenshot? What does the blue text on the RC phone read?

mihir-jain, 2020-02-09:
This is what happens what I run it: https://photos.app.goo.gl/Ntyfbo3uG1PCCNvb7
The blue text on the RC says: "Dashboard: server attached, no connections"

rbrott, 2020-02-09:
Is anything blocking the websocket? I would check the devtools network tab.

mihir-jain, 2020-02-09:
what do you mean?

rbrott, 2020-02-09:
https://developer.mozilla.org/en-US/docs/Glossary/Developer_Tools

mihir-jain, 2020-02-09:
I dont think so, because FTC Dashboard has worked before, just on skystone master, not on RR

mihir-jain, 2020-02-09:
And how do I check the devtools network tab?

mihir-jain, 2020-02-09:
[image]
It is doing this every few seconds, creating a new one

closed 2020-03-18 by mihir-jain