acmerobotics/road-runner-quickstart #61: Tuning StraightTest

jeevan-navudu, 2020-02-14:
I am stuck at tuning StraightTest.

[image]

Graph for StraightTest is looking like above. Telemetry data is 

Telemetry
heading: 6.190439185193185
headingError: 0.08762351641157018
mode: FOLLOW_TRAJECTORY
x: 91.65801460222416
xError: -31.052256029265354
y: -2.9784663244825853
yError: 0.2173338080607765

As this is straight test i only care about "x".  I have validated the robot position on the field and the position reported is right (There is one issue here too...) The robot still continued to move after the StraightTest is complete by 5 inches. I was logging the data that is how I concluded on that.

* What is validated - localizer is working well "X" reporting is perfect.
Robot configuration - 
1. I have threewheellocalizer and localizer is good at least from the straight test perspective
2. Drive train motors encoders are not connected and so not being used. set the RUN_USING_ENCODER to false.
3. Using 1150 RPM motors with external gear ratio lowered by half. Created new motortype and set highRPM to 1150 RPM. 
4. Ran the FeedForwardTuner thrice and used the average of the three runs 
```
    // First trial 0.01231, Second trial 0.01220, Third trial 0.01226 Avg = 0.01226
    public static double kV = 0.01226;
    // First trial 0.13538, Second trial 0.13155, Third trial 0.13246, Avg = 0.13313
    public static double kStatic = 0.13313;
    // First trial 0.00031, Second trial 0.00027, Third trial 0.00018, Avg = 0.00026
    public static double kA = 0.00026;

```
5. Initial values while running the FeedForwardTuner are 
`maxVel = 118.53, finalVel = 82.97, accel = 34.42
`

As the localizer is giving the values in inches, other than RPM of the motor I believe nothing else in DRIVE_CONSTANTS has an impact on the DriveFeedForwardTuner. 
I understand kV, kA, and kStatic are influencing the straight test, but not sure how to tune those.

What am i doing wrong?

rbrott, 2020-02-14:
What constraints are you using? Does the problem remain if you lower the constraints (i.e., lower maximum velocity, maximum acceleration)?

jeevan-navudu, 2020-02-15:
I was using the default constraints of maxVel 30 and maxAccel 30. I was getting an error 0f 50%. I reduced it to maxVel 10 and maxAccel 10 the error became 100%. Increased it to maxVel 80 and maxAccel 100 error is 0%.

[image]

I was expecting a flat line of error close to zero, instead the error is peaking till the midway and then coming down. Is this how the graph should be?

rbrott, 2020-02-15:
> I was getting an error 0f 50%. I reduced it to maxVel 10 and maxAccel 10 the error became 100%. Increased it to maxVel 80 and maxAccel 100 error is 0%.

This doesn't really make sense. I'm also unsure how to interpret the error percentages. Could you provide more specific information on the error? Like I ran the straight test with __ max vel, __ max accel for a distance of __ in and the final x error was __ in. 

Also you don't have to vary the drive constraints that extremely. I would stick to halving/doubling at most with the neighborhood of a multiplicative factor of 1.5 as a reasonable start.

> I was expecting a flat line of error close to zero, instead the error is peaking till the midway and then coming down. Is this how the graph should be?

No, that is not desirable. While the transient behavior isn't super important for the straight test, it is important for curved motion. You will also probably find that the feedforward parameters perform poorly with shorter/longer distances.

jeevan-navudu, 2020-02-16:
> This doesn't really make sense. I'm also unsure how to interpret the error percentages. Could you provide more specific information on the error? Like I ran the straight test with __ max vel, __ max accel for a distance of __ in and the final x error was __ in.

I ran the straight test with  30 max vel, 30 max accel for a distance of 60 inch and final x error was -23 inches. The results are pasted below.

Telemetry kV = 0.01226, kStatic = 0.13313, kA = 0.00026
heading: 0.22334560306261775
headingError: -0.22318383657078122
mode: FOLLOW_TRAJECTORY
x: 83.13107997683397
xError: -23.084156218350515
y: 4.969359398272146
yError: 0.2727692446022063

> Also you don't have to vary the drive constraints that extremely. I would stick to halving/doubling at most with the neighborhood of a multiplicative factor of 1.5 as a reasonable start.

Did that, first I lowered 30 max vel and 30 max accel to 20 max vel and 20 max accel. The 60 inches test ended with error of nearly -30 inch

Next step I lowered it to 10 max vel and 10 max accel. The 60 inches test ended with error of nearly -60 inches.

Then looked into the logs of DriveFeedforwardTuner and saw the velocities are around 60 and acceleration around 800. So tried 60 max vel and 60 max accel for 60 inches test. I ended with an error of 4 inches. Tried 80 max vel and 100 max accel for 60 inches test. I ended with a near 0 error. I tried the same settings for 40 inches test, I ended with 2 inches error.

The motors used are 1150 RPM with the external gear ratio of 0.5. This gives the maxVel as 118.53. So let me know if 80 max Vel and 100 max accel are not the right values I should be looking at?

> No, that is not desirable. While the transient behavior isn't super important for the straight test, it is important for curved motion. You will also probably find that the feedforward parameters perform poorly with shorter/longer distances.
I am confused now. As I am not using encoders on the motors (using three free wheels) so I am dependent completely on feedforward parameters. Right? If these do not perform well for shorter/longer distance, auton would not work well. Right?

rbrott, 2020-02-19:
> So let me know if 80 max Vel and 100 max accel are not the right values I should be looking at?

Lower constraints are nice for debugging because you can easily see what the robot is doing, the wheels will likely slip less (not a big issue with robust external localization), and you avoid plant nonlinearities that are exacerbated by high current (e.g., voltage sag). The main issue is not the choice of constraints, per se, but how much the accuracy varies with the constraints. To address this, you should revisit the feedforward tuner. In cases like this, it helps to look at the raw data to verify the relationship is linear (and if not find the range of velocities/accelerations where the relationship is linear). Also be very careful with kA; if the robot ever moves backward then kA is too high.

> As I am not using encoders on the motors (using three free wheels) so I am dependent completely on feedforward parameters. Right?

You're not entirely dependent on the feedforward response, although it should be doing a good amount of work (at the very least you don't want the feedforward to fight the feedback). 

> If these do not perform well for shorter/longer distance, auton would not work well. Right?

Correct, that's what we're trying to fix.

closed 2020-02-24 by jeevan-navudu