acmerobotics/road-runner-quickstart #62: Couldn't find Velocity Vs Power graph in Dashboard

jeevan-navudu, 2020-02-14:
Documentation on the page https://acme-robotics.gitbook.io/road-runner/quickstart/tuning says

> To find kV and kStatic, the robot executes a quasi-static ramp test where the power is slowly ramped up to minimize acceleration (it's effectively zero). Throughout this procedure, the velocity and power are recorded. In the corresponding velocity vs. power graph, kV is the slope and kStatic is the y-intercept. Next, to find kA, the robot attempts to accelerate rapidly from rest. This time, the acceleration, velocity, and power are recorded. The velocity is used to determine the acceleration-only power. The acceleration is then graphed against this new power, and the resulting slope is kA.

How do I view velocity vs power graph and acceleration vs power graph?

rbrott, 2020-02-14:
The cannot. The dashboard graphing features are designed for signals that evolve over time. 

The regression is performed automatically by the tuning routine. If you want to perform a manual regression, the data collected is stored in `/sdcard/RoadRunner`.

jeevan-navudu, 2020-02-15:
Your response answers my question. Thank you. Closing the ticket

closed 2020-02-15 by jeevan-navudu