acmerobotics/road-runner-quickstart #68: Accessing hardware parts în autonomous

Mihneaghitu, 2020-02-23:
Hello,
     We were building the autonomous while we encountered a problem: we cannot action every motor independently from our 'robot' object of the SampleMecanumRevOptimized class. For e ample, when we type ' robot. ' there seems to appear only the servo object we created in the hardware class. Could you provide a way to give power to the motor?
Thanks a lot!

rbrott, 2020-02-23:
If you want to access the motor instance variables externally, you can change their visibility to `public`.

Mihneaghitu, 2020-02-23:
Thanks a lot! Also, I am trying to load a file from `FtcRobotController/assets` to the method in the autonomous `TrajectoryLoader.load()`. How should the parameter in the method look like? My file is called trajectory_test.yaml.

rbrott, 2020-02-23:
"trajectory_test"

You should move the file into a `trajectory` folder within the assets folder.

Mihneaghitu, 2020-02-24:
I did both and it gives me an error: 
`Trajectory traj1 = TrajectoryLoader.load("trajectory_test");`
`load(java.io.file) in TrajectoryLoader cannot be applied to (java.lang.String).`

rbrott, 2020-02-24:
No, you should use `AssetsTrajectoryLoader`.

closed 2020-03-14 by rbrott