acmerobotics/road-runner-quickstart #69: FollowerPIDTuner and TRANSLATIONAL_PID

jeevan-navudu, 2020-02-24:
Robot : 

- Mechanum drive train, with GoBuilda 5220 1150 RPM motors
- drive train motors encoders are not connected to Rev Expansion Hub
- Standard three-wheel localizer is used, with these three wheels connected to Rev Expansion Hub

Drive Constraints set to 50 Max Vel, 100 Max Accel, 0 Max Jerk
Tuned kI, kA and kStatic to 
```
    public static double kV = 0.00970;
    public static double kStatic = 0.10346;
    public static double kA = 0.00044;

```
If Run straight test with the above constraints the graph looks like below
[image]

If I change TRANSLATIONAL_PID from all zeros to 

`public static PIDCoefficients TRANSLATIONAL_PID = new PIDCoefficients(5.0, 0.5, 0.0001);
`

The error goes crazy ...
[image]

I was hoping PID to help in error correction. Tried with only Proportional error set to 5.0 and others set to 0. Didn't make a big difference.

Could you let me know if I could expect a graph where the error is closer to zero during most of the timeline? If so what is the guidance to achieve it?

rbrott, 2020-02-25:
There's probably something wrong with your drive configuration. You should run `LocalizationTest` with and without the tracking wheel localizer and make sure the values are sane.

> Could you let me know if I could expect a graph where the error is closer to zero during most of the timeline?

Yes, that is the goal. Fixing the drive parameters will hopefully get you most of the way there.

> If so what is the guidance to achieve it?

For now, fix the feedforward by fixing the drive parameters.

jeevan-navudu, 2020-02-25:
> There's probably something wrong with your drive configuration. You should run LocalizationTest with and without the tracking wheel localizer and make sure the values are sane.

It's possible I am doing something wrong. Trying to figure out what it is... Your help is appreciated...
If I do not use the tracking wheel localizer, I have to connect the encoders on the drive motors. Right?

> For now, fix the feedforward by fixing the drive parameters.
I have been trying to do it. Initially, I was running things at default values. As the motors, I am using are high rpm motors 1150 RPM. the final velocity was too high. So I added a reduction factor of 0.85 and configured max power to 0.5 that reduced the max Velocity from 80 to 50. Not sure if that reduction is enough or if I have to go lower.

Look at my calculation - 1150 RPM - external gear reduction 0.5 - Wheel radius 2 inches 

1150 * 0.85 * 0.5 * 2 * 2 * Pi / 60 = approx 50 inches / sec

I have done FeedForwardTuner test on this. 
[DriveAccelRegression-Trial-1.csv.txt](https://github.com/acmerobotics/road-runner-quickstart/files/4248230/DriveAccelRegression-Trial-1.csv.txt)
[DriveAccelRegression-Trial-2.csv.txt](https://github.com/acmerobotics/road-runner-quickstart/files/4248231/DriveAccelRegression-Trial-2.csv.txt)
[DriveRampRegression-Trial-1.csv.txt](https://github.com/acmerobotics/road-runner-quickstart/files/4248232/DriveRampRegression-Trial-1.csv.txt)
[DriveRampRegression-Trial-2.csv.txt](https://github.com/acmerobotics/road-runner-quickstart/files/4248233/DriveRampRegression-Trial-2.csv.txt)

I see a max velocity of 50 and max acceleration of 700+ in the raw data... Is this sane data? I have added logging to logcat, for tracking the wheel positions and the wheel powers that are being applied. Can logging cause issues of the kind I am seeing?

ftc-team-astro-bruins, 2020-02-25:
Invited you to the repo. If you have few minutes to check and see if I am doing something wrong...

rbrott, 2020-02-25:
Alright I took a look at the data you provided and computing the following feedforward parameters:
kV = 0.0077
kStatic = 0.1641
kA = 0.00023  (this is likely too small to have a significant impact)

jeevan-navudu, 2020-02-25:
Thank you for your comments and sincerely appreciate the response I am receiving. Is there a process you followed to compute these values? I am asking because of the same raw data gave the tuning values differently. For example

The Trial-1 produced 
//    kV = 0.00968, kStatic = 0.10643 (R^2 = 0.93)
//    kA = 0.00033 (R^2 = 0.09)

The Trial-2 produced
//    kV = 0.00965, kStatic = 0.10582 (R^2 = 0.93)
//    kA = 0.00046 (R^2 = 0.13)

If I can understand then I would be able to apply that knowledge for further tests I am performing.

jeevan-navudu, 2020-02-25:
Tried the values you recommended

    public static double kV = 0.00770;
    public static double kStatic = 0.1641;
    public static double kA = 0.00023;

This resulted for 60 inch test an error of 9 inches. Error is still similar like before
You can see in the graph below 
[image]

[image]

rbrott, 2020-02-25:
> Is there a process you followed to compute these values?

I manually ran the regressions on the data you provided. When there is a lot of friction in the system, it helps to manually eliminate the low-voltage outliers before running the regression. 

I'm pretty confident that kV and kStatic are correct. I would try adjusting kA if I were you.

jeevan-navudu, 2020-02-26:
Thank you for the comments. Will try doing that and let you know.

jeevan-navudu, 2020-02-27:
After repeated failures with DriveFeedForwardTuner. I decided to try my luck with DriveVelocityPIDTuner.

Reconfigured the robot connected drive motor encoders to Expansion hub, set RUN_USING_ENCODER to true. After this, when I run DriveVelocityPIDTuner I see the following graph
Ran with default constraints MaxVel = 30, MaxAcel = 30, MaxJerk = 0
kP = 10, kI = 3, kD = 0

[image]

Started with kP = 2 kI = 0.5 kD = 0 and tried settings up to kP = 20, kI = 3, kD = 0. The graph isn't much different. I understand the graph would be different for every robot, but is there an ideal graph that I am optimizing for? Is this graph also as insane as the other graph I posted in this chat for StaightLine test?

Also are there any sane kP, kI and kD ranges that I should be playing with? Right now I am keeping kD at zero and trying to focus only on kP and kI

Your comments are appreciated.

jeevan-navudu, 2020-02-27:
After some more experimenting, I am going with kP = 40, kI = 0.9, kD = 0

Graph of that is looking as below.

[image]

closed 2020-02-29 by jeevan-navudu