acmerobotics/road-runner-quickstart #70: TrajectoryBuilder.reverse()

jeevan-navudu, 2020-02-27:
the reverse method was handy and the old SplineTest method was doing the reverse and tracing the path backward. Instead of the robot going in reverse it is taking a 180-degree turn and going crazy.

How do I achieve the same function as going through a spline and exactly tracing backward or reversing the robot direction?

rbrott, 2020-02-27:
Some of the `TrajectoryBuilder` constructor permit a `reversed` parameter.

jeevan-navudu, 2020-02-28:
Trajectory traj = drive.trajectoryBuilder(new Pose2d())
                .splineTo(new Pose2d(30, 30, 0))
                .build();

        drive.followTrajectory(traj);

        sleep(2000);

        drive.followTrajectory(
                drive.trajectoryBuilder(traj.end(), true)
                        .splineTo(new Pose2d(0, 0, 0))
                        .build()
        );

The above code does not do what I expect it to do. The forward spline is perfect. Then it sleeps for 2 seconds that is perfect too. After that robot takes one and a half rotation (360 + 180) and splines to the initial point and then takes another 180-degree turn. Apologies for asking basic questions.. The old test was perfect, it would be better if the new test does exactly what the old one was doing

jeevan-navudu, 2020-02-28:
Would you even recommend using roadrunner version 0.5.0 for the comp on March 7th? Or I should use version 0.4.7?

rbrott, 2020-02-28:
Have you updated your quickstart?

> Would you even recommend using roadrunner version 0.5.0 for the comp on March 7th? Or I should use version 0.4.7?

I'm not sure I can answer that for you. I would weigh the value of the 0.5.0 changes against the inevitable instability of new major releases.

jeevan-navudu, 2020-02-28:
> Have you updated your quickstart?
Yes. I have checked out a fresh copy of quick start few days ago..

rbrott, 2020-02-28:
> Yes. I have checked out a fresh copy of a quick start a few days ago..

I guess I'm confused because the latest quickstart doesn't have a method with signature `SampleMecanumDrive#trajectoryBuilder(Pose2d, boolean)`.

jeevan-navudu, 2020-02-29:
I have added that method as you mentioned i have to use the reversed parameter of Trajectory builder.

rbrott, 2020-03-02:
`SplineTest` has now been patched for 0.5.0 in https://github.com/acmerobotics/road-runner-quickstart/commit/6b25214a78360b9adebe23daa73648d8061f5e8d.

closed 2020-03-14 by rbrott