acmerobotics/road-runner-quickstart #71: Default Localizer Always Used

beaksnbolts, 2020-02-29:
We have added dead wheel encoders, and in our SampleMecanumDrive class we have added the following line in the identified location in the code:

    public SampleMecanumDriveBNB(HardwareMap hardwareMap) {
        super(kV, kA, kStatic, TRACK_WIDTH);
        setLocalizer(new StandardTrackingWheelLocalizer(hardwareMap));
//rest of the code

However all of our tests still seem to be relying on the drive encoders and the IMU.
Are we supposed to add/delete/modifying code anywhere else in order to use the 3 dead wheel localizer?

The only time the standardtrackingwheellocalizer seems to be used is during the localization test. The telemetry returned their for x,y, and heading comes from the dead wheels.

When we run turn test, changing the parameters in standardlocalizer doesn't affect the turn nor does changing those parameters affect the straight test or any of the other tuning test(track width, spline etc). Each of these test seems to rely on the drive wheel encoders and IMU.
Any help would be greatly appreciated. Thank you.

rbrott, 2020-02-29:
> Are we supposed to add/delete/modifying code anywhere else in order to use the 3 dead wheel localizer?

No, that's sufficient. You may want to the AS search or "find usages" feature to make sure there are no other conflicting `setLocalizer()` calls.

> When we run turn test, changing the parameters in standardlocalizer doesn't affect the turn nor does changing those parameters affect the straight test or any of the other tuning test(track width, spline etc).

Without follower PID, the localization doesn't affect the trajectory following. In this situation, the robot's motion is determined entirely by the drive/feedforward parameters and the trajectory.

beaksnbolts, 2020-02-29:
Thank you for your response. So if translational_pid and heading_pid are still set to 0, calling the 3 dead wheel localizer would have no effect and it would revert to the default localizer? We left translation_pid and heading_pid 0 because when we ran the pidfollower tuner the robot made pretty perfect squares the first couple times around, but again in retrospect it was only using the drive encoders and IMU. Thank you again.

rbrott, 2020-02-29:
> So if translational_pid and heading_pid are still set to 0, calling the 3 dead wheel localizer would have no effect and it would revert to the default localizer?

Yes, the choice of localizer will not affect the trajectory following (these two subsystems are independent without feedback). No, it will not revert to the default localizer.

beaksnbolts, 2020-02-29:
Maybe this is clarifying things or perhaps we are even more confused. So the spline test will always use the default localizer? What would we have to do in order to use the dead wheels to run a spline?

rbrott, 2020-02-29:
> So the spline test will always use the default localizer?

No. **Without feedback, the trajectory follower will not use localization at all.** The code will still respect your choice of localizer per usual, but **you shouldn't expect changing the localizer to affect the trajectory following.**

> What would we have to do in order to use the dead wheels to run a spline?

If you want the follower to use the localization information, start tuning the PID gains.

beaksnbolts, 2020-02-29:
thank you! We will stay working at it. Sharing your time and knowledge with the FTC community is greatly appreciated. Thank you.

closed 2020-02-29 by beaksnbolts