acmerobotics/road-runner-quickstart #72: Exceptions in logs

jeevan-navudu, 2020-03-01:
I see the below exceptions in the logs. Need help on understanding what needs to be done to address these exceptions... I am referring to 

E/ClassManager: java.lang.ClassNotFoundException: kotlin.streams.jdk8.StreamsKt$asStream$1
        at java.lang.Class.classForName(Native Method)
        at java.lang.Class.forName(Class.java:324)

02/29 22:39:49: Launching 'TeamCode' on LGE Nexus 5.
$ adb shell am start -n "com.qualcomm.ftcrobotcontroller/org.firstinspires.ftc.robotcontroller.internal.PermissionValidatorWrapper" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Waiting for process to come online...
Connected to process 22795 on device 'lge-nexus_5-192.168.49.1:5555'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/cr_LibraryLoader: Loaded native library version number "80.0.3987.119"
W/chromium: [WARNING:dns_config_service_posix.cc(341)] Failed to read DnsConfig.
V/OnBotJava:ClassLoader: opening DexFile /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar
I/OnBotJavaHelperImpl: /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.ABotUtil
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig$ArmDirection
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig$FoundationServoState
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig$IntakeDirection
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig$LiftDirection
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig$SkyStonePosition
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig$strafeDirection
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.AstroBotHwConfig
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.DoubleDriver
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.MotorPowers
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.TestPlatformServos$1
    /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar: org.firstinspires.ftc.teamcode.TestPlatformServos
V/OnBotJava:ClassLoader: opening DexFile /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar
I/art: Rejecting re-init on previously-failed class java.lang.Class<kotlin.streams.jdk8.StreamsKt$asStream$1>
W/ClassManager: kotlin.streams.jdk8.StreamsKt$asStream$1 java.lang.ClassNotFoundException: kotlin.streams.jdk8.StreamsKt$asStream$1
E/ClassManager: java.lang.ClassNotFoundException: kotlin.streams.jdk8.StreamsKt$asStream$1
        at java.lang.Class.classForName(Native Method)
        at java.lang.Class.forName(Class.java:324)
E/ClassManager:     at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.classNamesToClasses(ClassManager.java:206)
        at org.firstinspires.ftc.robotcore.internal.opmode.ClassManager.processAllClasses(ClassManager.java:255)
        at com.qualcomm.ftccommon.ClassManagerFactory.processAllClasses(ClassManagerFactory.java:90)
E/ClassManager:     at org.firstinspires.ftc.robotcontroller.internal.FtcRobotControllerActivity.onCreate(FtcRobotControllerActivity.java:335)
        at android.app.Activity.performCreate(Activity.java:6251)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1107)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2369)
E/ClassManager:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2476)
        at android.app.ActivityThread.-wrap11(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1344)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:148)
E/ClassManager:     at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
E/ClassManager: Caused by: java.lang.ClassNotFoundException: kotlin.streams.jdk8.StreamsKt$asStream$1
E/ClassManager:     at org.firstinspires.ftc.onbotjava.OnBotJavaClassLoader.findClass(OnBotJavaClassLoader.java:211)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
        at org.firstinspires.ftc.onbotjava.OnBotJavaClassLoader.loadClass(OnBotJavaClassLoader.java:201)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:469)
    	... 18 more
E/ClassManager: 	Suppressed: java.lang.ClassNotFoundException: Didn't find class "kotlin.streams.jdk8.StreamsKt$asStream$1" on path: DexPathList[[zip file "/data/app/com.qualcomm.ftcrobotcontroller-2/base.apk"],nativeLibraryDirectories=[/data/app/com.qualcomm.ftcrobotcontroller-2/lib/arm, /data/app/com.qualcomm.ftcrobotcontroller-2/base.apk!/lib/armeabi-v7a, /vendor/lib, /system/lib]]
E/ClassManager:     at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:511)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
    		... 20 more
E/ClassManager: 		Suppressed: java.lang.NoClassDefFoundError: kotlin.streams.jdk8.StreamsKt$asStream$1
        at dalvik.system.DexFile.defineClassNative(Native Method)
        at dalvik.system.DexFile.defineClass(DexFile.java:226)
E/ClassManager:     at dalvik.system.DexFile.loadClassBinaryName(DexFile.java:219)
        at dalvik.system.DexPathList.findClass(DexPathList.java:338)
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:54)
    			... 22 more
    		Suppressed: java.lang.ClassNotFoundException: kotlin.streams.jdk8.StreamsKt$asStream$1
E/ClassManager:     at java.lang.Class.classForName(Native Method)
        at java.lang.BootClassLoader.findClass(ClassLoader.java:781)
        at java.lang.BootClassLoader.loadClass(ClassLoader.java:841)
E/ClassManager:     at java.lang.ClassLoader.loadClass(ClassLoader.java:504)
    			... 21 more
    		Caused by: java.lang.NoClassDefFoundError: Class not found using the boot class loader; no stack trace available
V/OnBotJava:ClassLoader: From /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar loaded: org.firstinspires.ftc.teamcode.ABotUtil
    From /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar loaded: org.firstinspires.ftc.teamcode.AstroBotHwConfig$ArmDirection
V/OnBotJava:ClassLoader: From /storage/emulated/0/FIRST/java/build/jars/onBotJavaJar-2020-02-29+22.03.50-902007205/OnBotJava.jar loaded: org.firstinspires.ftc.teamcode.AstroBotHwConfig$FoundationServoState

rbrott, 2020-03-01:
Is the exception fatal?

jeevan-navudu, 2020-03-02:
Not sure. App isn't crashing when the exception occurs. But I am not sure if this causes an issue either in functionality or performance later on.
What I was expecting is one of the following
1. This is an exception that authors are aware of, and I should not worry about it.
2. There is a build configuration I missed that I should fix to get rid of this warning. If so what is the configuration?
3. If I have to upgrade the SDK version I am using, or something similar.

rbrott, 2020-03-02:
I'm not aware of this error, but I think you can safely ignore it.

jeevan-navudu, 2020-03-04:
Thank you for your response

closed 2020-03-04 by jeevan-navudu