acmerobotics/road-runner-quickstart #74: Velocity PID questions

Unggoy117, 2020-03-08:
Hi Ryan,
This is Aurick, I'm learning your PID controller, but need help with understanding how to tune the velocity pid. I ran the VelocityPIDTuner but none of the changes I make to the coefficients seem to have any effect.  I have attached three screen shots of three graphs, the first with the default coefficients, the second with increased P, and the third with increased I.  As you can see the graphs look almost identical.  Am I missing something basic? 
[image]
[image]
[image]

Thank you in advance for your help

rbrott, 2020-03-08:
> As you can see the graphs look almost identical. Am I missing something basic?

Typically the coefficients are larger (especially kP). You may want to start with a P of around 20-25.

Unggoy117, 2020-03-09:
Hey thanks for the help, I had someone tell me that the values were very sensitive so I was confused when nothing seamed to change. I just did these tests with the P as 27 and the I as 1. It seemed to fix my first problem quite well but now I have this new problem ware the motors are moving to quickly, do you think this is o.k. PS I tried lowering the P to 24 and just got the worst of both Wolds, (spikiness, and lag).
Thanks again in advance, Aurick

rbrott, 2020-03-09:
> I just did these tests with the P as 27 and the I as 1. It seemed to fix my first problem quite well but now I have this new problem ware the motors are moving to quickly, do you think this is o.k. PS I tried lowering the P to 24 and just got the worst of both Wolds, (spikiness, and lag).

Do you have a screenshot of the response? The main goal of the tuning process is to balance oscillations and phase lag. I suggest you play around and get an intuition for how the coefficients affect the graphical response.

Unggoy117, 2020-03-09:
Hey, I tried to reply to your message with a picture of the new graph using Email but it didn't go through. Can you see it here?
[image]
Also I am starting to get an intuition for the variables through tinkering with them, I just was hoping you could tell me if these spikes were normal, especially when the motors are supost to have reached the zero power mark. Over all I just don't have a good idea of what the graph should look like when I'm done, can you look at the graph and tell me what I should work on next. It would be great if I could see an example of a graph from what you consider a well tuned velocity PID.  Do you have an example?

rbrott, 2020-03-10:
> I just was hoping you could tell me if these spikes were normal, especially when the motors are supost to have reached the zero power mark.

Those oscillations do seem a bit too intense. 

> It would be great if I could see an example of a graph from what you consider a well tuned velocity PID. Do you have an example?

Sure

[image]

closed 2020-05-23 by rbrott