acmerobotics/road-runner-quickstart #76: RoadRunner Crashes

San68bot, 2020-03-19:
Hi,

When I try doing either a spline/line constant heading then a regular spline, the RC app just crashes and I have to restart the app, is there a way I can fix this? 

drive.followTrajectory(
                drive.trajectoryBuilder(new Pose2d())
                        .splineToConstantHeading(new Pose2d(0, -37))
                        .splineTo(new Pose2d(40, -32, 0.0))
                        .build()
        );

Thanks,
Sanjay!

rbrott, 2020-03-20:
The trajectory you provided seems continuous. Do you have any logs or a stacktrace with the specific error?

San68bot, 2020-03-20:
It's not showing an error but when I run it, it crashes the app

San68bot, 2020-03-20:
I also tried the same thing with an older version of RR quick start and it works fine

closed 2020-03-24 by San68bot