acmerobotics/road-runner-quickstart #80: using the odometry wheels to self correct

ZLDman, 2020-06-22:
How do you set the odometry wheels (unpowered) to self correct while following a trajectory?
So if it gets off a little or hits something it will end up at the proper place.

rbrott, 2020-06-22:
In short, you create a subclass of `TwoTrackingWheelLocalizer` or `ThreeTrackingWheelLocalizer` and call `setLocalizer(new MyLocalizer(...))` at the end of your `Drive` subclass constructor. This is briefly covered in [the docs](https://acme-robotics.gitbook.io/road-runner/tour/kinematics#tracking-wheels).

ZLDman, 2020-06-22:
I have three tracking wheels and there have been very accurate. But when I try to follow a trajectory sometimes the robot will get a little off especially when strafing. Is the only way to get more accurate results to adjust the feedforward parameters?

rbrott, 2020-06-22:
> But when I try to follow a trajectory sometimes the robot will get a little off especially when strafing.

Have you tuned the parameters of the follower? That is, if you're using a PIDVA follower, have you tuned the PID coefficients?

> Is the only way to get more accurate results to adjust the feedforward parameters?

No, you can use the feedback afforded by your follower of choice. You can, however, use feedforward tweaks to ease the feedback burden. For example, if you're using the mecanum drive classes/kinematics, you can adjust the `lateralMultiplier` (you can replace [this line](https://github.com/acmerobotics/road-runner-quickstart/blob/84b81c888893ea6f321c26382bc18330b6687ee6/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/SampleMecanumDrive.java#L83) with `super(kV, kA, kStatic, TRACK_WIDTH, TRACK_WIDTH, LATERAL_MULTIPLIER);`).

ZLDman, 2020-07-30:
How do I set up the 
> feedback afforded by your follower of choice.

rbrott, 2020-07-30:
The details depend on the follower. For example, if you're using the quickstart mecanum, you can adjust the values on [these lines](https://github.com/acmerobotics/road-runner-quickstart/blob/7778ecc9f0b5515aed1244d2e1d32a0f6618aefe/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/SampleMecanumDrive.java#L54-L55). These parameters are passed to the follower during [construction](https://github.com/acmerobotics/road-runner-quickstart/blob/7778ecc9f0b5515aed1244d2e1d32a0f6618aefe/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/SampleMecanumDrive.java#L96).

closed 2020-12-16 by rbrott