acmerobotics/road-runner-quickstart #81: [Q] Drive simulator based on roadrunner

wangxdflight, 2020-07-01:
We are trying to develop a driver simulator, as there is need to training/learning without access to physical robot under pandemic.

It seems we just need to implement virtual motors, virtual drive train, where motor velocity can be simulated and returned based on input power settings, if we don't have to simulate too much details.

Not sure if there is known similar effort.

mabeeenigma, 2020-07-01:
You could look at this project:
https://github.com/Beta8397/virtual_robot and see if it fits your needs. It
already incorporates RR core.

wangxdflight, 2020-07-02:
Thank you. Will take a look.

shaunsingh, 2020-10-12:
I'm not sure if it is different from what's above, but there is also https://github.com/Beta8397/virtual_robot_rr from the same team which is dedicated to Road Runner

rbrott, 2020-12-16:
The core library also has tests like [`MecanumFollowerTest`](https://github.com/acmerobotics/road-runner/blob/master/core/src/integration-test/kotlin/com/acmerobotics/roadrunner/MecanumFollowerTest.kt) that use primitive simulation.

closed 2020-12-16 by rbrott