acmerobotics/road-runner-quickstart #82: Problem with StandardTrackingWheelLocalizer and LocalizationTest

foocross, 2020-07-13:
We have adapted StandardTrackingWheelLocalizer and filled out DriveConstants and ran LocalizationTest. LocalizationTest gives incorrect heading. x and y. The x and y cycle to -12 and then back up again to 0 when we drive the robot forward. We have tested the math behind the localization - even implementing the trig functions ourselves and our calculations seem correct.

Below are our drive constants and StandardTrackingWheelLocalizer. Any suggestions would be greatly appreciated :)

`
package` org.firstinspires.ftc.teamcode.drive;

import com.acmerobotics.dashboard.config.Config;
import com.acmerobotics.roadrunner.control.PIDCoefficients;
import com.acmerobotics.roadrunner.trajectory.constraints.DriveConstraints;

@Config
public class DriveConstants {

    /*
     * These are motor constants that should be listed online for your motors.
     */
    public static final double TICKS_PER_REV = 537.6;
    public static final double MAX_RPM = 312;

    /*
     * Set the first flag appropriately. If using the built-in motor velocity PID, update
     * MOTOR_VELO_PID with the tuned coefficients from DriveVelocityPIDTuner.
     */
    public static final boolean RUN_USING_ENCODER = false;
    public static final PIDCoefficients MOTOR_VELO_PID = null;

    /*
     * These are physical constants that can be determined from your robot (including the track
     * width; it will be tune empirically later although a rough estimate is important). Users are
     * free to chose whichever linear distance unit they would like so long as it is consistently
     * used. The default values were selected with inches in mind. Road runner uses radians for
     * angular distances although most angular parameters are wrapped in Math.toRadians() for
     * convenience. Make sure to exclude any gear ratio included in MOTOR_CONFIG from GEAR_RATIO.
     */
    public static double WHEEL_RADIUS = 1.9685; // in
    public static double GEAR_RATIO = 1.1538461538461537; // output (wheel) speed / input (motor) speed
    public static double TRACK_WIDTH = 12.5; // in
    /*
     * These are the feedforward parameters used to model the drive motor behavior. If you are using
     * the built-in velocity PID, *these values are fine as is*. However, if you do not have drive
     * motor encoders or have elected not to use them for velocity control, these values should be
     * empirically tuned.
     */
    public static double kV = 0.01;
    public static double kA = 0.00001;
    public static double kStatic = 0.23;
    /*
     * These values are used to generate the trajectories for you robot. To ensure proper operation,
     * the constraints should never exceed ~80% of the robot's actual capabilities. While Road
     * Runner is designed to enable faster autonomous motion, it is a good idea for testing to start
     * small and gradually increase them later after everything is working. The velocity and
     * acceleration values are required, and the jerk values are optional (setting a jerk of 0.0
     * forces acceleration-limited profiling). All distance units are inches.
     */
    /*
     * Note from LearnRoadRunner.com:
     * The velocity and acceleration constraints were calculated based on the following equation:
     * ((MAX_RPM / 60) * GEAR_RATIO * WHEEL_RADIUS * 2 * Math.PI) * 0.8
     * Resulting in 59.36856133047848 in/s.
     * This is only 80% of the theoretical maximum velocity of the bot, following the recommendation above.
     * This is capped at 80% because there are a number of variables that will prevent your bot from actually
     * reaching this maximum velocity: voltage dropping over the game, bot weight, general mechanical inefficiences, etc.
     * However, you can push this higher yourself if you'd like. Perhaps raise it to 90-95% of the theoretically
     * max velocity. The theoreticaly maximum velocity is 74.21070166309809 in/s.
     * Just make sure that your bot can actually reach this maximum velocity. Path following will be detrimentally
     * affected if it is aiming for a velocity not actually possible.
     *
     * The maximum acceleration is somewhat arbitrary and it is recommended that you tweak this yourself based on
     * actual testing. Just set it at a reasonable value and keep increasing until your path following starts
     * to degrade. As of now, it simply mirrors the velocity results in 59.36856133047848 in/s/s
     *
     * Maximum Angular Velocity is calculated as maximum velocity / (trackWidth / 2) * (180 / Math.PI)
     */
    public static DriveConstraints BASE_CONSTRAINTS = new DriveConstraints(
            59.36856133047848, 59.36856133047848, 0.0,
            Math.toRadians(566.928), Math.toRadians(566.928), 0.0
    );

    public static double encoderTicksToInches(double ticks) {
        return WHEEL_RADIUS * 2 * Math.PI * GEAR_RATIO * ticks / TICKS_PER_REV;
    }

    public static double rpmToVelocity(double rpm) {
        return rpm * GEAR_RATIO * 2 * Math.PI * WHEEL_RADIUS / 60.0;
    }

    public static double getMotorVelocityF() {
        // see https://docs.google.com/document/d/1tyWrXDfMidwYyP_5H4mZyVgaEswhOC35gvdmP-V-5hA/edit#heading=h.61g9ixenznbx
        return 32767 * 60.0 / (MAX_RPM * TICKS_PER_REV);
    }
}
`

'
package org.firstinspires.ftc.teamcode.drive;

import android.support.annotation.NonNull;

import com.acmerobotics.dashboard.config.Config;
import com.acmerobotics.roadrunner.geometry.Pose2d;
import com.acmerobotics.roadrunner.localization.ThreeTrackingWheelLocalizer;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.HardwareMap;

import java.util.Arrays;
import java.util.List;

/*
 * Sample tracking wheel localizer implementation assuming the standard configuration:
 *
 *    /--------------\
 *    |     ____     |
 *    |     ----     |
 *    | ||        || |
 *    | ||        || |
 *    |              |
 *    |              |
 *    \--------------/
 *
 */
@Config
public class StandardTrackingWheelLocalizer extends ThreeTrackingWheelLocalizer {
    public static double TICKS_PER_REV = 2400;
    //public static double WHEEL_RADIUS = 1.1811; // in
    public static double WHEEL_RADIUS = 1.1811; // in
    public static double GEAR_RATIO = 1.0; // output (wheel) speed / input (encoder) speed

    public static double LATERAL_DISTANCE = 14.15; // in; distance between the left and right wheels
    public static double FORWARD_OFFSET = -6.5; // in; offset of the lateral wheel

    private DcMotor leftEncoder, rightEncoder, frontEncoder;

    private int leftEncoderMod = 1;
    private int rightEncoderMod = 1;
    private int frontEncoderMod = 1;

    public StandardTrackingWheelLocalizer(HardwareMap hardwareMap) {
        super(Arrays.asList(
                new Pose2d(.5, LATERAL_DISTANCE / 2, 0), // left
                new Pose2d(.5, -LATERAL_DISTANCE / 2, 0), // right
                new Pose2d(FORWARD_OFFSET, 0, Math.toRadians(90)) // front
        ));

        leftEncoder = hardwareMap.dcMotor.get("back_right");
        rightEncoder = hardwareMap.dcMotor.get("front_right");
        frontEncoder = hardwareMap.dcMotor.get("back_left");

    }

    public double getLeftEncoder() {
        return encoderTicksToInches(leftEncoder.getCurrentPosition()*leftEncoderMod);
    }
    public double getRightEncoder() {
        return encoderTicksToInches(rightEncoder.getCurrentPosition()*rightEncoderMod);
    }
    public double getFrontEncoder() {
        return encoderTicksToInches(frontEncoder.getCurrentPosition()*frontEncoderMod);
    }

    public static double encoderTicksToInches(int ticks) {
        return WHEEL_RADIUS * 2 * Math.PI * GEAR_RATIO * ticks / TICKS_PER_REV;
    }

    @NonNull
    @Override
    public List<Double> getWheelPositions() {
        return Arrays.asList(
                encoderTicksToInches(leftEncoder.getCurrentPosition()*leftEncoderMod),
                encoderTicksToInches(rightEncoder.getCurrentPosition()*rightEncoderMod),
                encoderTicksToInches(frontEncoder.getCurrentPosition()*frontEncoderMod)
        );
    }
}
'

rbrott, 2020-07-13:
I don't see any obvious red flags. I would suggest double-checking the wheel coordinates (check out [this example](https://acme-robotics.gitbook.io/road-runner/tour/kinematics#tracking-wheels)). In particular, the `frontEncoder` position is currently behind the robot center.

foocross, 2020-07-13:
The left and right encoders are slightly (~.5 inches) forward of the center robot and the lateral is slightly off from the center. We can try adjusting the parameters. Would that make a difference? Also does it matter that the track width of the encoder wheels are larger than the drive wheels?

rbrott, 2020-07-13:
> Would that make a difference?

Actually, the wheels can translate in their orientation direction without changing the kinematics. The slight forward offset of the left and right encoders doesn't factor in; the lateral distance controlling. Nevertheless, I recommend you put in a value anyway but don't stress about fine tuning it.

> Also does it matter that the track width of the encoder wheels are larger than the drive wheels?

That's perfectly fine.

If the wheel positions seem correct, you may have an encoder direction problem. You can reverse direction by changing the angle (e.g., 0 rad to pi rad) or negating the encoder reading.

ftcteam8645, 2020-08-01:
In your SampleMechanumDrive Class, do you have any motors reversed?  We had this same issue and had to change the heading for the encoder that was plugged into the reversed motor by 180 degrees.  It is kind of odd, but this fixed our issue.

foocross, 2020-08-04:
Hi ftcteam8645 - I'm not sure I understand what you mean by reversing the motor by 180. Can you elaborate?

ftcteam8645, 2020-08-04:
Something like this:
        leftFront = hardwareMap.get(DcMotorEx.class, "lfd");
        leftRear = hardwareMap.get(DcMotorEx.class, "lrd");
        rightRear = hardwareMap.get(DcMotorEx.class, "rrd");
        rightFront = hardwareMap.get(DcMotorEx.class, "rfd");

        // TODO: reverse any motors using DcMotor.setDirection()
        leftFront.setDirection(DcMotorSimple.Direction.REVERSE);
        leftRear.setDirection(DcMotorSimple.Direction.REVERSE);

If you have the motor reversed and then assign one of these 'Reversed motor ports' to your dead wheel encoder, it will report negative values.

        leftEncoder = hardwareMap.dcMotor.get("lrd"); // THIS ONE WILL REPORT NEGATIVE VALUES DUE TO BEING ASSOCIATED WITH THE REVERSED MOTOR PORT.

        rightEncoder = hardwareMap.dcMotor.get("rrd");
        frontEncoder = hardwareMap.dcMotor.get("rfd");

It seems you must 'fix' this by changing the heading of that encoder

                new Pose2d(0, LATERAL_DISTANCE / 2, 0), // left  <<< This one will need to be Math.toRadians(180) if not flipped or 0 if flipped

                new Pose2d(0, -LATERAL_DISTANCE / 2, 0), // right
                new Pose2d(FORWARD_OFFSET, 0, Math.toRadians(90)) // front

foocross, 2020-08-08:
Thanks we will try this.

closed 2020-12-16 by rbrott

sharkree, 2021-03-26:
> The x and y cycle to -12 and then back up again to 0 when we drive the robot forward

To whom ran into this again, your direction setup could be wrong. If you do this as we did,
```
   // TODO: reverse any encoders using Encoder.setDirection(Encoder.Direction.REVERSE)
    leftEncoder.setDirection(Encoder.Direction.REVERSE);
```
and the X cycles through when you drive forward, you shall comment it out or reverse both encoders. For example,
```
        // TODO: reverse any encoders using Encoder.setDirection(Encoder.Direction.REVERSE)
        leftEncoder.setDirection(Encoder.Direction.REVERSE);
        rightEncoder.setDirection(Encoder.Direction.REVERSE);
```

If X is negative when you drive forward, you can simply remove both statements.