acmerobotics/road-runner-quickstart #91: StraightTest and Dashboard

rischconsulting, 2020-09-15:
We've got an odd tuning problem, and can't find the answer in the docs. If we're missing it, please feel free to point me there. We've tuned Kv,Ka,Kstatic (though we think not well - doing this at home and we don't have enough runway to generate a plateau in target velocity. But it's close.

We tried straight test, and we find that it very consistently hits whatever distance we give it, but not in the way we expect - the FRONT of the robot goes 60 inches, rather than the back or center. It's as if the system knows we have an 18" bot and that we want to drive the front the appropriate distance from the rear.  

We see two things on the dashboard - a blue circle and a green circle. We also see an x that is much less than 60in and an xerror that is 17 or so. We suspect it's not working as intended, but we can't find any documentation on what the two colors means or what xerror means. Any assistance would be much appreciated.

Thanks!

rbrott, 2020-09-15:
> We tried straight test, and we find that it very consistently hits whatever distance we give it, but not in the way we expect - the FRONT of the robot goes 60 inches, rather than the back or center.

The robot should travel 60in; it shouldn't matter whether you use the front or back as a reference.

> We see two things on the dashboard - a blue circle and a green circle. 

Blue circle = estimated pose; where the robot thinks it is based on localization/encoder readings
Green circle = desired pose; where the robot should be according to the trajectory and current time

> We also see an x that is much less than 60in and an xerror that is 17 or so.

This seems consistent with the robot's actual movement on the field; the localization appears to be working. The value `x` is the x-coordinate of the estimated pose in the field frame, and the value `xError` is the x-coordinate deviation between the desired pose and the estimated pose in the robot frame (here the robot and field frames are aligned). 

Given this, it seems like the feedforward values are a bit off. Can you share some graphs of desired and measured velocity over time?

rischconsulting, 2020-09-20:
Thanks for the quick reply. We solved the problem, but for posterity, I’ll report the solution. The issue was that MaxVel and MaxAccel were WAY too high – each 146. They were set by the learnroadrunner.com wizard, and based on our ridculously fast drivetrain (3.7:1 NeverRest with 2:1 bevel gears). This is why our expected velocity never plateaued – we hit the 100 inches well before we reached peak. We rammed the walls quite a few times before we figured it out, and even when we didn’t the robot slid all over the field it was going so fast. As a result, the straight test overshot all the time even when we thought it was tuned, unless  we tuned it way way down, in which case it never got there.

So, we reduce maxVel and maxAccel to more reasonable values – like 80 or 90, and that allowed us to properly tune and all worked well.

We ran into the same issues with maxAngVel, which was set to something like 10 radians (1000+ degrees). In this case, our turn test turned 22.5 degrees instead of 90, and also overshot and came back to that. The tuning opmode suggested a trackwidth of .25. We reduced angular velocity to 2.7 radians and it worked fine.

Regards,
MR

rbrott, 2020-09-20:
Thanks for the update. That is a very fast drivetrain.

CC @NoahBres

NoahBres, 2020-09-20:
Apologies for the trouble! I do not think the constant generator for LRR has a problem for the estimated max velo.  Considering you're going 800 RPM, the math does check out. I do know that I need to lower angular velocity considerably (the calculation for it is incredibly naive) but considering your 800 RPM drive train it would result in large numbers :P That also just depends on the track width you specified.

I don't know what your drive train looks like but I'm assuming the generated constants simply do not work with your current configuration. Perhaps it doesn't have enough torque to reach the specified acceleration (given the very low gear ratio). Can't really tell. But such high velocities is not an edge case I thought about. My apologies. I will definitely cap the max velocity on the generator. And make an addendum in LRR.

> This is why our expected velocity never plateaued – we hit the 100 inches well before we reached peak.
That's not really a problem with anything. You just didn't have enough distance to make the entire trapezoidal motion profile. 

> but we can't find any documentation on what the two colors means or what xerror means
LRR does actually specify what these are, just in a later step. I did not think they would be necessary for the FF step. There's also a note in the Velocity PID Tuning page about lowering the max velocity as a trouble shooting step but I did not think to put that in the Feed Forward page. My deepest apologies for the inconvenience. I really appreciate the feedback. 

Thank you so much!

NoahBres, 2020-09-20:
Just out of curiosity, what form of odometry are you using that stays accurate at such high speeds? I'm assuming standard 3-wheel dead wheels?

rischconsulting, 2020-09-20:
Thanks for the feedback! We cut video of our robot outrunning competition to get skystones for scouting last year.

And we are loving the LRR, which has been extraordinarily helpful, especially for the kids who are on the young side but picking this up quickly.

The math is definitely correct from a calculation point of view. We’re using the GoBilda strafer kit, just with swapped out motors. So, we’re pretty confident on track width, etc. But you’re right – torque is a bit of an issue, but the bigger issue was just friction (or lack thereof). We had wheels spinning freely and robot sliding around on both startup and slowdown – so we could never get the slopes to align for Ka, nor could we get a plateau for Kv.

It’s working well now – we just did the closed loop tuning and we are shocked at how perfect the square is (we’re going to do the alternate test to really get the PID tuned exactly next).

rischconsulting, 2020-09-20:
We did the 3 dead wheels design from Wizards.exe – so US Robotics e8t with a 1.5in (or so, it’s metric) omni in the spaces of the chassis.

I have no idea if they were accurate at that speed. Maybe that was part of the problem! Our straight test never got to that speed even when we had the higher maxVel. The only time it ever got to such a high speed was the manual tuning, and it could be that our numbers didn’t tune right because our measured velocity was wrong due to odometry issues (in addition to sliding around  when braking and wheel spin).

Regards,
MR

NoahBres, 2020-09-20:
> And we are loving the LRR, which has been extraordinarily helpful, especially for the kids who are on the young side but picking this up quickly.

Thank you! I am trying my best to make it accessible to a wide audience (thus the use of prose everywhere :P) but if there's any wording that's confusing to them let me know! I am constantly changing LRR with feedback, trying to clarify as much as I can. 

> torque is a bit of an issue, but the bigger issue was just friction (or lack thereof). We had wheels spinning freely and robot sliding around on both startup and slowdown – so we could never get the slopes to align for Ka, nor could we get a plateau for Kv.

Ah. I'm sure that becomes a massive problem at high velocities. Reducing acceleration significantly in the `DriveConstants` should definitely help with that.

> We did the 3 dead wheels design from Wizards.exe

Gotcha. Just a warning but I did use their design and had a pretty poor experience with them. The design doesn't really have anything that retains the shaft so when the shaft moved, the encoder disc slid up and down with it, dragging the disc, scratching it, and rendering it useless. I spent too much on replacement discs. Hopefully y'all won't experience such problems but if you do, I do recommend the [Open Odo design](https://openodometry.weebly.com/) as it does utilize a through bore encoder. But it's a pretty pricey investment so if you don't have problems with the wizard's design then there's no reason to move from it.

> I have no idea if they were accurate at that speed

Dead wheels will definitely experience slip if not properly sprung. I have seen some teams experience inaccuracies in dead wheel measurements when their bot goes faster than around 50 in/s if they only spring with a single rubber band. I personally used 3 rubber bands per dead wheel and then switched to springs and found it pretty reliable. But be wary of poor springing affecting the dead wheels. I believe the wizard's tutorial videos only call for a single rubber band but I believe that isn't quite enough force to be accurate at medium velocities.

> It’s working well now

Wonderful! Best of luck!

rischconsulting, 2020-09-20:
Thanks – we’ll look at the Open Odo. We use springs, and they are pretty tight – indeed, the kids want to move to a pod-style vertical design, because they are tired of having to reset these wheels in the right direction because the pull is so strong. We found the same shaft problem with the wizards design, which is one reason we didn’t go live with it last year. This summer, we experimented and found that jamming a plastic spacer in held the shaft in place better, and are getting good enough results for proof of concept and initial path planning while the new robot is being designed.

Regards,
MR

NoahBres, 2020-09-24:
> because they are tired of having to reset these wheels in the right direction because the pull is so strong

If you stick a little spacer or nub (we used a small screw and standoff) that essentially stops the dead wheels from rotating past 90 degrees then you won't have to reset the wheels in a certain orientation. You should be able to lift up the bot and the standoffs prevent the wheels from flipping into the other direction.

rischconsulting, 2020-09-24:
Ok, so we need meepmeep install help. We'll reach out.

Regards,
MR (from mobile)

NoahBres, 2020-09-25:
MeepMeep is currently under development and I haven't published installation instructions because it relies on something called TrajectorySequence which has not yet been finalized. I'll let you know when that's finished :). For now, I would recommend the two other path visualization alternatives.

rischconsulting, 2020-12-13:
Different robot, similar problem. We're on to this year's robot. Same drive motors, but using chain rather than bevel for same gear ratio. Wheels still spin crazily (we'll have to cut back on the teleop power, as full joystick is just too much). Using rev encoders in a spring pod design. Localization test seems accurate -- x and y increase when we drag it, but heading is off by 40% ish when we turn it by hand. We haven't gotten to turn test to fix our track width (maybe it's off?)

We've tuned Kv, Ka, Kstatic. The slopes and peaks are OK, but we can't get the actual to plateau- unclear why. Maybe not enough acceleration?

We're at the straight and strafe test in learn road runner's order.. When we got straight, it consistently stops at 48" and then strafes left by about 5". We're scratching our head over that one. When we strafe test, it goes in the proper direction, but nowhere near straight - it sort of curves in a way that the wheel power is not right. Odder still is that the dashboard thinks the robot moved in the opposite direction (which implies a front encorder reversed, but I don't think so - we'll check it).

Any thoughts appreciated - especially as to why the robot might suddenly strafe at the end of drive straight.

rischconsulting, 2020-12-15:
nevermind my  last post - lateral_distance issues

closed 2020-12-15 by rischconsulting