acmerobotics/road-runner-quickstart #100: [Q] how combine RR with vision alignment for target

adamrubinfeld, 2020-10-14:
hi
for this year (and every year with shooting) u need to alignment with vision in auto and teleop
this year we choose to use RR because it is very easy and reliable 
how can I combine the RR and vision alignment??

tnx for helpers

EddieDL, 2020-10-24:
There are a number of different approaches to using vision to accomplish a task with RoadRunner. The simplest is to use vision to check what the physical position of something relative to where your robot currently is. Then move your robot to a given pose in response to that.

This is what many teams did last year.
1. Move to a known position where you should see a Stone
2. Check if you see a SkyStone.
3. Move to grab the SkyStone.

However, in this years game the targets don't change position. So using vision to acquire the goals may not be necessary or practical (there isn't a vision target on any of the goals). You may be able to simply use RoadRunner to move to a known pose and shoot. 

What you will need vision for is to understand where to place the Wobble goal.

adamrubinfeld, 2020-10-24:
ok tnx

closed 2020-10-24 by adamrubinfeld