acmerobotics/road-runner-quickstart #105: [Q] How do you actually use MaxVelocityTuner?

EddieDL, 2020-10-23:
@NoahBres I think the MaxVelocityTuner is a great addition. I just am not sure how it is really supposed to be used.

[This](https://github.com/acmerobotics/road-runner-quickstart/commit/dd266178aa0a26ee552907a0bd769aed8026ba4c#diff-7764e6660298cbd4e47126dc62b9b50779ed6b0ae4606789258042806aa9e180R17) documentation isn't very clear about how to use the outputs of this opmode. Based on the code, I think what a user would do is update the following line in DriveConstants.java `public static PIDFCoefficients MOTOR_VELO_PID = new PIDFCoefficients(0, 0, 0, getMotorVelocityF(MAX_RPM / 60 * TICKS_PER_REV));` to use the voltage compensated kF? But maybe it should just be the raw kF value? Can you confirm what you are supposed to actually do here?

NoahBres, 2020-10-23:
You simply replace the `getMotorVelocityF()` part with whatever number the tuner outputs.
It should be voltage compensated kF because the `setPIDFCoefficients` function in `SampleMecanum/TankDrive` scales the kF value based on current voltage so it works well across the entire range of your battery life (you can see that a low battery at 12.5v will require a higher kV than one at 13.5v). But `MaxVelocityTuner` calculates the raw kF so we have to apply an inverted voltage compensation for it to be a simple plug and play.

Would the label be less confusing if telemetry simply printed "kF" instead of "Voltage Compensated kF"?

EddieDL, 2020-10-24:
Now that I look at the math in all of the different places a bit closer and I think about what `setPIDFCoefficents` is doing this makes perfect sense.

I do think just calling it "kF" would be make this much easier. I also think some comments on the lines/files where this is being done would help too. It isn't necessarily glaring that you are doing `effectiveKf * batteryVoltageSensor.getVoltage() / 12` in one and `coefficients.f * 12 / batteryVoltageSensor.getVoltage()` in the other. Also, I never thought to check the `setPIDFCoefficents` method for the other conversion.

Thank you very much for the quick response.

NoahBres, 2020-10-24:
I'll make a PR renaming it to "kF." I do agree that the "voltage compensation" part may lead to confusion without context. 
I didn't add any comment explanations regarding the details of the voltage compensation because the rest of the quickstart doesn't really have any comments. Apologies if the implementation isn't obvious.

I would have to get Ryan's input to see if he comments would be beneficial before adding them. 
@rbrott

EddieDL, 2020-10-24:
No need to apologize you are doing great work. 

I get being consistent with the rest of the code base. What if you just add an expiration to the top if the opcode to let the user know what to do with the output and maybe something about the meaning of voltage compensation. 

I could write something if you want as well.

EddieDL, 2020-10-24:
I also noticed that you added documentation for this to the learn road runner site. That helps a lot, I was trying it before that was updated I think.

rbrott, 2020-10-25:
> I would have to get Ryan's input to see if he thinks comments would be beneficial before adding them.

I don't think the comments are necessary, but I'm fine with adding them. Once these op modes prove themselves, the information can find a home in the official docs.

EddieDL, 2020-10-25:
Sounds good. I really think the addition in the Learn Road Runner helps a ton.  I was a bit "bleeding edge" :P

closed 2020-10-25 by EddieDL

NoahBres, 2020-10-25:
> Sounds good. I really think the addition in the Learn Road Runner helps a ton.  I was a bit "bleeding edge" :P

You definitely were considering I added the explanation to LRR like less than 24 hours after it was merged :P