acmerobotics/road-runner-quickstart #106: "Event Loop Detached" Error In FTC Dashboard

rahulravi1026, 2020-10-24:
@NoahBres 

When we run the updated code with MaxVelocityTuner on the REV Control Hub, we are receiving an error saying "event loop detached" on the FTC Dashboard while the old project still works. By any chance, do you know why this is the case and how we can fix it? Thank you!

NoahBres, 2020-10-24:
Does your FtcRobotControllerActivity have all the necessary code for Dashboard? Detailed [here](https://acmerobotics.github.io/ftc-dashboard/gettingstarted)?

rahulravi1026, 2020-10-24:
Thanks for the quick response! After going through the guide, our FtcRobotControllerActivity does have all the necessary code for the Dashboard. What's interesting is the outdated Roadrunner project (from a couple of weeks ago) still works on the Dashboard. When our team realized that new code had been committed for the MaxVelocityTuner, we downloaded the project once again, and the Dashboard returned the "event loop detached error".

NoahBres, 2020-10-28:
Are you still encountering the issue? The new code hasn't touched anything regarding dashboard. Ryan may be better suited for such a problem.

rbrott, 2020-10-28:
This problem tends to occur when starting the app via hardware attachment (similar to https://github.com/acmerobotics/ftc-dashboard/issues/11). This startup process bypasses the normal callbacks and consequently `attachEventLoop()` is never called. Many features still work in this case (e.g., telemetry, config variables, camera stream).

I'm not sure if this problem exists on the CH. The only possibility I can think of at the moment is the watchdog killed the app due to misbehaving user code and a hardware event somehow opened the app before the normal notification process. In any case, I imagine restarting the CH would resolve the issue.

rahulravi1026, 2020-10-28:
We are still encountering the issue. We will do our best to restart the Control Hub in our next meeting on Friday and will update you on how it goes. Thank you!

closed 2020-12-16 by rbrott