acmerobotics/road-runner-quickstart #108: [Q] Does Road Runner use the Odometry Wheel Velocity?

EddieDL, 2020-10-28:
Our team is looking at using the Open Odometry design for our dead wheels. One concern with this design is that it uses the REV Through Bore encoder. This encoder has a super high resolution, which we think would be fine for localization. However, as we understand it the `getVelocity()` method uses 16 bit integers which means it won't correctly monitor the velocity at "high" speed. Below is an issue from last year where this was discussed.

This raises our question of if Road Runner actually uses the velocity from the dead axle wheels if we are using the motor velocity PID? If it isn't used we think we could safely use this design for odometry wheels. If it is used, we will need to determine if we will be impacted based our potential velocity.

[Issue](https://github.com/FIRST-Tech-Challenge/SkyStone/issues/241#issue-571667374)

rbrott, 2020-10-28:
The quickstart does use the velocity from the dead axle wheels, and the issue you described is mitigated by https://github.com/acmerobotics/road-runner-quickstart/pull/90. You shouldn't have to worry if you use `Encoder`.

EddieDL, 2020-10-28:
That fix seems cool, but it doesn't look to me like the change actually uses the `getCorrectedVelocity()`. I feel like to really fix this we need to change from using `getRawVelocity()`. Is that right?

rbrott, 2020-10-28:
Yes, that's correct. I should have made that more clear before. The feature is opt-in because it might misbehave (although it's only used for tuning at this point).

NoahBres, 2020-10-29:
[image]

EddieDL, 2020-10-29:
@NoahBres  I feel dumb, I was looking right at that code and apparently completely ignored the wonderful comment.

We currently don't use the Rev robotics encoders or the omni wheels from the Open Odometry design. So we would have to invest to convert to this design. Our current implementation doesn't exceed 32767 counts / second. However, our current implementation is MUCH bigger than the Open Odometry design.

The only pause I have in making the investment now is @rbrott's statement below. I reviewed the code and it seems pretty solid. Should we be concerned about it misbehaving or is it just a possibility?

> because it might misbehave

NoahBres, 2020-10-29:
I believe the misbehaving comment was precautionary? The correction is kind of a hack, albeit a really neat one.

rbrott, 2020-10-29:
> Should we be concerned about it misbehaving or is it just a possibility?

You probably shouldn't be very concerned. Malfunction is a slim possibility. However, if you know the correction is never applicable for your operating speed and resolution, it's safer to use the velocity directly instead of relying on the correction to be a no-op.

closed 2020-11-10 by EddieDL