acmerobotics/road-runner-quickstart #109: Maximum achievable accuracy with odometry

odpgit, 2020-11-01:
When testing a robot using REV through-bore encoders as an odometry system, we noticed a significant drift in the y-dimension despite the fact that we were just running the robot in a straight line on the x-axis. Testing further, we put two encoders on the same shaft (pictured below). We modified StandardTrackingWheelLocalizer to treat the bonded encoders as left and right while always returning 0 for the front, perpendicular encoder. When spinning the shaft with the two encoders on it, we noticed some difference between the two encoder counts and a significant drift in the y-axis over time. For example, using LocalizationTest with our bonded encoders over the equivalent of 100 feet of back and forth motion, there would be a drift of 2 inches in the y-dimension, which clearly is impossible.

We experimented with different bulk caching modes, and while we got the best result with manual cache control, we still had 2+ inches of phantom y drift. 

The question is whether this is simply the inherent limitation of the physical and electrical system or if there is something else to look into.

[image]

NoahBres, 2020-11-02:
> we noticed some difference between the two encoder counts

If the two encoder counts don't match, the issue is probably that the heading drifts ever so slightly. The heading is calculated via the difference between the two parallel wheels so discrepancies between those two encoders would cause the heading to drift so the bot isn't exactly straight and you would get slight movement in the Y direction. You should try printing the heading and I am betting you will see that.

Seems like a sensor issue from what you've described. Although 2 inches over 100ft is a 0.001% so I think that's well within REV's expected tolerances. Not completely sure why that's a concern.

NoahBres, 2020-11-02:
If you do notice slight drift in the y direction, you might want to consider seeing if your heading is off during normal use. Might just be a need to further fine tune your track width. Or maybe Ryan might have some further insight.

Tangentially related issue but the dead wheel setup seems to experience a slight movement in the x direction when you strafe the bot back and forth. It's very slight but noticeable when you simply rock the bot back and forth with no x displacement. Seems to be an issue with omni wheels in general. Ethan from goBILDA has an explanation as to why although I do not recall it off the top of my head.

I don't think the drift you've demonstrated can be avoided though. Other than binning REV encoders to match drift patterns.

odpgit, 2020-11-21:
I'm just getting back to testing this now. I still think there is something wrong. In the latest test run with a modified version of LocalizationTest, the same experimental setup showed a phantom y drift of 0.5 inches over 93 inches of x travel. This is a 0.5% error, but more importantly it doesn't seem reasonable that odometry with this accuracy of encoder should drift as much as 0.5 inches in less than 8 feet. Remember that this is with a fixed shaft that is keeping both encoders turning at exactly the same rate for the same distance, so this is a best case result. Attaching to independent omni wheels on a robot will likely have a larger error. I would have thought that using precision encoders in an odometry setup would a better result. Are others really not getting better accuracy than this?

Note that the localizer does show a nonzero heading in the output below, so the localizer does think the heading's changed due to the discrepancy between the encoder values over time.

Is it possible that this is caused by the lag between reading the two encoders' values inside the Control Hub? Obviously, the two values can't be physically read at exactly the same time, so could this discrepancy cause the localizer to accumulate error over time? 

[image]

rbrott, 2020-11-22:
> I would have thought that using precision encoders in an odometry setup would a better result. Are others really not getting better accuracy than this?

Other setups anecdotally achieve better accuracy, although I haven't heard of real systematic testing.

> Is it possible that this is caused by the lag between reading the two encoders' values inside the Control Hub? Obviously, the two values can't be physically read at exactly the same time, so could this discrepancy cause the localizer to accumulate error over time?

Lag is somewhat plausible if you aren't making use of bulk reads, although the latest version of the quickstart does (unless you somehow have the encoders on different hubs).

EddieDL, 2020-11-22:
@odpgit last year we had significant issues, similar to what you are describing. We didn't use the REV encoders, but another high resolution encoder (cross the road's mag encoders) which have 4096 counts per rev. Back then, we didn't even thinking about the "fix" that is currently implemented. 

This season we were considering going to the REV encoders, since there are open source designs for them. Instead we went to "old school" encoders with I think 256 (might be 128) counts per rev. What I can tell you is that we haven't yet had any of the issues we had last year.

I would say absolutely that the encoder counts were an issue, but I am fairly confident that you are better off sticking to something that is lower resolution.

Regarding the lag you mentioned and @rbrott commented on. Any possible lag could be picked up with the higher resolution encoders, where the lower resolution it would be much less likely. At least in theory.

odpgit, 2020-11-22:
@rbrott We are already using bulk reads through the manual bulk caching mode. The lag that I was wondering is the lag between the successive reads that I assume that the Control Hub is doing to the encoder hardware.

@EddieDL Thanks for the suggestion. I'll experiment with lower resolution encoders and see how they work.

rbrott, 2020-11-22:
In that case I doubt there's material lag between the signals.

However, it's somewhat easy to check if you record the raw encoder position time series. The localizer is only obscuring the results.

closed 2021-12-15 by rbrott