acmerobotics/road-runner-quickstart #110: [Q] What are the rules for odometry wheel location?

EddieDL, 2020-11-10:
I am not sure if I am missing this in the docs or if it is supposed to just be obvious. What are the rules/requirements for the location of odometry wheels?  Specifically, things like:

1. Do the two parallel wheels need to be "in-line"? For example, could one wheel be 1 inch forward of the other wheel?
2. Does the perpendicular wheel need to be in the center of the robot or could it be offset right or left?
3. Does the center of the "lateral distance" need to be in the center of the robot? 

All of the pictures I found show the parallel wheels in-line and the perpendicular wheel in the center of the robot. However, in the following code, it seems like you could easily just adjust the following code to replace the X or Y values that are currently 0 to be the offset in the other direction. We could also adjust the LATERAL_DISTANCE value so that it isn't centered either.

I guess I am asking if ese are a requirement or just best practice/easiest? Would the math still work right?

```
        super(Arrays.asList(
                new Pose2d(0, LATERAL_DISTANCE / 2, 0), // left
                new Pose2d(0, -LATERAL_DISTANCE / 2, 0), // right
                new Pose2d(FORWARD_OFFSET, 0, Math.toRadians(90)) // front
        ));
```

EddieDL, 2020-11-10:
Another similar question.

4. Can any wheels be in the "exact" center of rotation?

rbrott, 2020-11-10:
> What are the rules/requirements for the location of odometry wheels?

There are no rules or requirements so long as the configuration is not degenerate. For example, you cannot determine strafe movement with three parallel wheels. If you're mathematically inclined, degeneracy here means the matrix described in section 3.2 of [this doc](https://github.com/acmerobotics/road-runner/blob/master/doc/pdf/Mobile_Robot_Kinematics_for_FTC.pdf) is singular. As usual, one should also prefer a well-conditioned matrix. However, this isn't really an issue in practice. 

> All of the pictures I found show the parallel wheels in-line and the perpendicular wheel in the center of the robot. However, in the following code, it seems like you could easily just adjust the [...]

While it's typical to have a configuration like that (hence the "standard" moniker), the `ThreeTrackingWheelLocalizer` interface and implementation impose not constraint beyond non-degeneracy. 

> I guess I am asking if ese are a requirement or just best practice/easiest? Would the math still work right?

This appears to be common practice. Personally, it seems better to orient wheels in the principal directions of travel and place them for easy measurement. However, I'm utterly unqualified to comment on the physical construction and mounting of these wheels. I can only confirm that the math "works" subject to the aforementioned constraints.

> Can any wheels be in the "exact" center of rotation?

Yes, that's fine.

EddieDL, 2020-11-11:
@rbrott The math in section 3.2 is a bit over my head. However, I am going to try to ask some questions to see if I can understand better (please be kind :)). Also, it is really long and sort of stream of thought :(.

I think this is the root equation you were talking about.
[image]
Then you created an example configuration like below:
[image]
This results in the following equation.
[image]

One of my questions is if you changed this configuration to the configuration below, what changes.
[image]

Based on my understanding of the formula, it has no impact what so ever. In my example, if the two parallel wheels are at
[image]. The x term has no bearing on the formula because it would still be multiplied by [image]. Which is 0 in our case.

In the case of the other wheel its still fine because we just changed the value of [image], Which is already accounted for in the formula.

If we further change the configuration as follows:
[image]

We still end up with the same equation. In this case the the value for the single wheel on the y axis [image]. Again it has no impact on the equation because [image] is 0.

This also would imply that the following configuration would also have no change as well (even though it seems unnatural to me).
[image]

Where it would seem you would need to change things is if you change the orientation of your odometry wheels or if you offset your two parallel wheels. For example the following configurations:
[image]
[image]

In the first scenario you just have to use the exact terms for y instead of [image].

In the second scenario the entire equation is different and you need to figure out if it will be degenerate as you described above.

Is my general understanding correct?  If so, I think the general rules I was looking for are:

1. Make sure your wheels are parallels to the axis of movement they are measuring.
2. Make sure your two parallel wheels are equidistance from the center axis they are measuring.

So long as 1 and 2 are adhered to you can put your wheels any where you want on the robot and the only thing the QuickStart cares about is the LATERAL_DISTANCE and FORWARD_OFFSET.

_Rule 2 isn't really required as you can easily manually update the y location for each of those wheels if you have to._

If it isn't clear through all of this, it isn't natural for me to believe that you only really care about the location of the wheel in the opposite axis from what it is measuring. Which appears to be the case, based on the formulas. Maybe this is obvious or documented somewhere, but it is something that has nagged at me for 2 seasons now :)

EddieDL, 2020-11-11:
Side note, using technology to write math is a problem for the ages. I am sure there is some better tool, but I can't imagine writing a paper like the one you wrote (and I am pretty sure that isn't your only one) :P.

rbrott, 2020-11-11:
> it isn't natural for me to believe that you only really care about the location of the wheel in the opposite axis from what it is measuring. Which appears to be the case, based on the formulas.

Yes, you've stumbled upon a somewhat surprising property of wheel placement. One can slide/translate a wheel in its orientation direction without changing the math. For this reason, the second, third, and fourth configurations you presented are all equivalent. The first configuration is different because the strafe/perpendicular wheel has been moved _perpendicular_ to its orientation direction.

> In the first scenario you just have to use the exact terms for y instead of [image].
> 
> In the second scenario the entire equation is different and you need to figure out if it will be degenerate as you described above.

Yes, both of those assessments are correct. 

> Is my general understanding correct? If so, I think the general rules I was looking for are:
> 
>     1. Make sure your wheels are parallels to the axis of movement they are measuring.
> 
>     2. Make sure your two parallel wheels are equidistance from the center axis they are measuring.

Well the easiest approach is to ignore the math and accurately take all position measurements. I think your general understanding is mostly correct, although neither of those rules is necessary. If you want to minimize the measurements required, you can translate all wheels into a "radial" configuration and just measure the distance from there to the origin (although that may be a more difficult measurement to take). 

> So long as 1 and 2 are adhered to you can put your wheels any where you want on the robot and the only thing the QuickStart cares about is the LATERAL_DISTANCE and FORWARD_OFFSET.

Yes, that's correct. All configurations with two parallel/axial wheels and one perpendicular/strafe/lateral wheel can be reduced to a canonical configuration of the `StandardTrackingWheelLocalizer` form. 

> Side note, using technology to write math is a problem for the ages.

Most definitely. I'm slowly becoming facile with LaTeX, but I have a long way to go.

closed 2020-11-11 by EddieDL