acmerobotics/road-runner-quickstart #111: Rev Encoder Spikes

Jaiming724, 2020-11-19:
[image]

Myself and 2 others in the ftc community have experience spikes while tuning feedforward with 3 wheel odometry. The 3 wheel odometry are using rev encoders, so I have reason to believe the spikes are caused by the wrapper when it tries to correct for the overflow. Possible solutions would be to either apply a rolling error on the velocity or just filter out large change of velocity all together. I am not sure what the best solution here would be, any input would be appreciate. Thank you!

rbrott, 2020-11-19:
Can you capture a file with the current timestamp, encoder position, and raw encoder velocity for one encoder over time?

Jaiming724, 2020-11-22:
http://0x0.st/iRP0.csv. I ran the manual feedforward tuner and had it write the data to a csv. The time is the profileTime so that is why it is resetting. Please let me know if you need any more data.

rbrott, 2020-11-22:
This data is a bit fishy. Consider the left encoder for the first profile. The effective velocity range appears to be ~2048 (0x800) instead of ~65536 (~10000) as expected for 16-bit ints.

Jaiming724, 2020-11-26:
Sorry for the late response. What would you suggest to do next?

rbrott, 2020-11-26:
Perhaps get data from one of the other teams experiencing this issue. Seems unlikely that you're collecting the wrong data, but I would double check that no post-processing is occurring.

Jaiming724, 2020-11-26:
This is a bit ugly, but I have just writing to a csv as the robot is running.
`            pw.println(profileTime+","+drive.getWheelPositions().get(0) + "," + drive.getWheelVelocities().get(0) + "," + drive.getWheelPositions().get(1) + "," + drive.getWheelVelocities().get(1)  + ","+ drive.getWheelPositions().get(2) + ","+ drive.getWheelVelocities().get(2) + ",");` Do you see an issue with this?

rbrott, 2020-11-26:
Oh yeah that's bad. You don't want the values to go through `encoderTicksToInches()`. Not sure why I missed that; for some reason I (mis)remembered that the values were integers in the CSV.

rbrott, 2020-11-27:
Just send your localizer file and I can back out the original ticks from the parameters.

Jaiming724, 2020-11-30:
https://hastebin.com/hesegepude.java This is our localizer. I reran it Sunday and all the spikes disappeared with no code changes. Someone also reported the same thing. I am not quite sure why. Should I close the issue until it starts happening again?

Jaiming724, 2020-11-30:
I misspoke, the other user used Heno's fix to remove the spikes. Yet I did not change anything to fix the spikes and it still disappeared.

rbrott, 2020-12-01:
I cannot reproduce spikes from the data you provided. I'd guess that your loop time is too slow to cause them. I'd still like to keep this issue open though if other people have experience this problem. I'd like data before merging #113.

Jaiming724, 2020-12-02:
I am currently writing to 4 motors and bulk reading once. I am not sure what else I can do to increase loop time.

rbrott, 2020-12-02:
Your loop time is ~60ms which is a bit on the high side in my experience (certainly for just reading/writing motors). The delay is probably due to your logging code; if you're using the auto bulk read mode, each call to `getWheelPositions()`/`getWheelVelocities()` will trigger a new bulk read.

Jaiming724, 2020-12-07:
How should I log without adding the long delay?

rbrott, 2020-12-08:
The easiest is probably to use manual bulk reads and clear the cache in the loop. Otherwise you can cache the values and add another set of accessors or even just move the logging logic into the localizer.

Jaiming724, 2020-12-08:
I will do that next time we meet, thank you!

rbrott, 2022-08-19:
Fixed in #113

closed 2022-08-19 by rbrott