acmerobotics/road-runner-quickstart #120: dashboard does not display graph on AutomaticForwardFeedTuner

barbaralau3, 2021-01-20:
Hi, I followed the instruction and ran AutomaticForwardFeedTuner with 
RegressionUtil.RampResult rampResult = RegressionUtil.fitRampData(
                timeSamples, encoderVel1Samples, encoderVel2Samples, fitIntercept,
                LoggingUtil.getLogFile(Misc.formatInvariant(
                        "DriveRampRegression-%d.csv", System.currentTimeMillis())));
       
These ArrayList (timeSamples...) has thousand items. However, the graph is not displaying anything. I clicked "graph" button after I init and start the program. I also cannot find the "DriveRampRegression-XXXX.csv" file.

I really appreciate your help! Thank you.

rbrott, 2021-01-20:
You shouldn't really expect useful items to graph for the automatic tuner. For better or worse, the automatic routines don't provide much user feedback besides the final values.

The files should be under `/sdcard/FIRST/RoadRunner`.

barbaralau3, 2021-01-20:
Thank you for your reply. I am actually testing the shooting wheel with two
dc motors. I hope any graph of time, velocities of two motors are
consistent after PID. Therefore the graphing function is important for now
and future. I will test tomorrow morning.

To be honest, I still don’t know why the graph is not showing. The
telemetry session in dashboard displays something. The variable
configuration is on right side. I will update with you tomorrow if I still
cannot retrieve / create the graph .

Thank you again,
Barbara

closed 2021-12-15 by rbrott