acmerobotics/road-runner-quickstart #121: debugging localizer heading

Team14423, 2021-01-26:
Hi - what might be the best way to debug our localizer heading? Some thoughts on what we've done:

1. x/y appear to be accurate at slow (drag) speed, and heading used to work at high speed. We will continue to work on this, but unclear how to best narrow it down to this since the heading change on the dash screws up the x/y readings. 
2. Did the Dec. code change affect anything that we might have missed in a merge?
3. We ran lateralwidthtuner. It showed 12.6", even though a) we can easily measure our omni wheels at 14" and b) the dashboard shows our robot heading perfectly over 10 turns. What accounts for a 10% error in cumulative heading but accurate current heading?
4. The dashboard stays pretty accurate even if we change a relatively high speed in the tuner, but NOT in localization test. Why might that be?
5. We have vx_weight and vy_weight way down in samplemechanumdrive - otherwise, the wheels spin way too fast. Even as it is, we cannot strafe straight - the robot pivots as we strafe, presumably due to weight imbalance. We'll figure that out, but it won't affect localization, will it?

Any input on how to track this down appreciated. Thanks!

rbrott, 2021-01-26:
>     1. x/y appear to be accurate at slow (drag) speed, and heading used to work at high speed. We will continue to work on this, but unclear how to best narrow it down to this since the heading change on the dash screws up the x/y readings.

I favor low speed as a better source of truth.

>     2. Did the Dec. code change affect anything that we might have missed in a merge?

I'm not exactly sure to which code change(s) you're referring, but I doubt you're missing any critical code.

>     3. We ran lateralwidthtuner. It showed 12.6", even though a) we can easily measure our omni wheels at 14" and b) the dashboard shows our robot heading perfectly over 10 turns. What accounts for a 10% error in cumulative heading but accurate current heading?

The error in the current heading can seem smaller than the true if it wraps around. You should pay more attention to cumulative heading. 

>     4. The dashboard stays pretty accurate even if we change a relatively high speed in the tuner, but NOT in localization test. Why might that be?

If you're still just considering final measurements, the issue I raised previously still applies. Try changing the total turn angle or look at the error over time.

>     5. We have vx_weight and vy_weight way down in samplemechanumdrive - otherwise, the wheels spin way too fast. Even as it is, we cannot strafe straight - the robot pivots as we strafe, presumably due to weight imbalance. We'll figure that out, but it won't affect localization, will it?

No, the weights do not affect localization.

Team14423, 2021-01-29:
Thanks. We think we may have fixed it. We put a camera under the bot and watched the wheels (which worked). We wound up manually tuning lateral_distance until the heading matched - 14" became 13.81" - we'll see if that works.

closed 2021-04-01 by Team14423