acmerobotics/road-runner-quickstart #124: Missing getHeadingVelocity method

odpgit, 2021-02-03:
When running the ManualFeedforwardTuner, the program threw an exception. The error message was "poseVelocity() must not be null. Ensure that the getWheelVelocities() method has been overridden in your localizer." It refers to ManualFeedforwardTuner.java:119. Using the debugger, we traced the issue to the heading velocities being null when the pose velocity is set. It seems like there should be a getHeadingVelocity() method in the localizer method (we use StandardTrackingWheelLocalizer) to correct this, but there isn't. getWheelVelocities() has been overridden, so that is not the issue. Is there a reference implementation for getHeadingVeolocity() somewhere or a standard way to implement it? We are using two wheel odometry with the built-in Rev Hub IMU.

rbrott, 2021-02-03:
Is it possible that the localizer wasn't set on the drive class? Maybe try logging `drive.getLocalizer().getClass().getName()` before `ManualFeedforwardTuner.java:119`.

odpgit, 2021-02-03:
The localizer is definitely set on the drive class, we've done other tests with it.

rbrott, 2021-02-03:
Hmmm well then I'm not sure what's going on. There's a pretty direct connection between `drive.getPoseVelocity()` and `localizer.getWheelVelocities()`.

odpgit, 2021-02-03:
getWheelVelocities() isn't the problem. It's getHeadingVelocity(). That is the one that is returning a null value.

odpgit, 2021-02-03:
val wheelVelocities = getWheelVelocities()
        val headingVelocity = getHeadingVelocity()
        if (wheelVelocities != null && headingVelocity != null) {
            poseVelocity = calculatePoseDelta(wheelVelocities, headingVelocity)
        }

This is the code (TwoTrackingWheelLocalizer.kt) that is causing poseVelocity to be null, because getHeadingVelocity() is returning null, the default value.

rbrott, 2021-02-03:
Ah the three tracking wheel localizer is a red herring. I think a sufficient implementation is `return imu.getAngularVelocity().firstAngle;`.

odpgit, 2021-02-03:
Thanks, we'll try that out!

BNGCNLR, 2021-02-04:
I had the same problem, and I solved it by writing this in SampleMecanumDrive.java:
    ```
@Override
    public Double getExternalHeadingVelocity(){
        return imu.getVelocity().xVeloc;
    }
```

and this in TwoWheelTrackingLocalizer.java:
  ```
  @NonNull
    @Override
    public Double getHeadingVelocity(){
        return Objects.requireNonNull(drive.getExternalHeadingVelocity(), "heading velocity is null");
    }
```

BNGCNLR, 2021-02-04:
I had the same problem, and I solved it by writing this in SampleMecanumDrive.java:

@Override
    public Double getExternalHeadingVelocity(){
        return imu.getVelocity().xVeloc;
    }

and this in TwoWheelTrackingLocalizer.java:

  @NonNull
  @Override
    public Double getHeadingVelocity(){
        return Objects.requireNonNull(drive.getExternalHeadingVelocity(), "heading velocity is null");
    }

rbrott, 2021-02-04:
This comment prompted me to actually look up the API docs. The `imu.getVelocity()` call returns integrated **translational** velocity, while the `imu.getAngularVelocity()` call returns the **angular** velocity. You want the latter, not the former. Looks like `AngularVelocity` doesn't have a `firstAngle` field, and I'm not exactly sure which rotation rate is correct.

odpgit, 2021-02-05:
Yes, we are using imu.getAngularVelocity() and assuming that the z rotation is the correct value to use, since logically the robot would rotate around the z-axis. Does that make sense?

rbrott, 2021-02-05:
Yes, Z-axis makes the most sense, but it depends on your sensor placement and remapping.

BNGCNLR, 2021-02-08:
Thanks for the clarification!

PR #127 (2021-02-16): Override getExternalHeadingVelocity
Fixes #124
closed 2021-02-16 by rbrott